bambulab/BambuStudio #1600: Reload from disk does not work with STP-files loaded as a "single object with multiple parts"

nervnet65, 2023-04-11:
The feature under Process Object "Reload from disk" leads to the error message "Unable to reload: filepath/filename(s)" when the object was loaded from multiple STP files ("Load these files as a single object with multiple parts? -> Yes). This error message appears both when trying to reload a single STP file or the entire object.

Reload from disk works when an obejct consists of only one single STP file.

If STL files are used, the error does not occur.

BambuStudio 1.5.0.61 unter Windows 10

zhimin-zeng-bambulab, 2023-04-26:
@nervnet65  Could you share your step files, it will help us to solve this problem.

nervnet65, 2023-04-26:
@zhimin-zeng-bambulab I have created a simple part and attach it as STL and STP. I was also able to successfully test the described behavior with it.
[object_STL_and_STP.zip](https://github.com/bambulab/BambuStudio/files/11329761/object_STL_and_STP.zip)