bambulab/BambuStudio #1601: Linux version crashed after sending to printer with segentation error

A-nglea, 2023-04-11:
Using Fedora 36. 
Bambu Studio crashed after negotiating cloud send of file to printer. Printer received file ok.

dmesg showed:-
[10850.281459] bambustu_main[7481]: segfault at 50 ip 0000000002f36d88 sp 00007fb1d37f4530 error 4 in bambu-studio[4a5000+3144000] likely on CPU 0 (core 0, socket 0)
[10850.281484] Code: 00 00 b8 01 00 00 00 48 81 c4 58 01 00 00 5b 5d 41 5c 41 5d 41 5e 41 5f c3 66 0f 1f 84 00 00 00 00 00 48 89 df e8 08 f9 56 fd <48> 39 68 50 75 1a 8b 50 40 83 fa 02 0f 84 f6 09 00 00 83 fa 40 0f

Happens off an on with no obvious trigger. Not sure of the steps to repeat but a file dropped on the prepare plate and sliced. No errors in slicing. Send to printer dialogue was fine and file sent. Bambu Studio then crashed.

lanewei120, 2023-04-13:
@A-nglea 
hello, thanks for reporting this issue
1. where is the studio from? built bu your self or downloaded from github
2.  it is it a 100% reproduciable issue or not?
3. which machine are you using?(X1C or P1P?)
4. are you using lan mode?

A-nglea, 2023-04-13:
Version is from Bambu Studio:-
Bambu_Studio_linux_fedora_v01.05.00.61_20230314200047.AppImage 
Fault is not repeatable, but random similar crashes have occurred. They are, however, few.
Using X1C
And not in LAN mode but using cloud for connectivity.

A-nglea, 2023-04-14:
Will try compiling...
Meanwhile another crash attempting to put text on a model.
dmesg:-
[80011.587211] bambustu_main[25049]: segfault at 39 ip 00007fa7ac34ac16 sp 00007ffd67dfadb0 error 4 in libstdc++.so.6.0.30[7fa7ac29a000+111000] likely on CPU 3 (core 3, socket 0)
[80011.587218] Code: 00 00 00 90 f3 0f 1e fa 48 39 f7 0f 84 93 00 00 00 41 56 4c 8d 77 10 41 55 41 54 55 48 89 f5 53 48 89 fb 48 83 ec 10 4c 8b 2f <4c> 8b 66 08 4d 39 ee 74 79 48 8b 57 10 4c 39 e2 72 20 4d 85 e4 75

lanewei120, 2023-06-10:
@A-nglea do you still have this issue?

there is a guide on the wiki page
https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide for linux compiling

A-nglea, 2023-06-10:
Updates seem to have cleared the issue, thanks . Impressed by the follow-up though! :)

closed 2023-06-12 by lanewei120