bambulab/BambuStudio #1607: Prevent "Objects" to auto-drop on the table

bolsoncerrado, 2023-04-13:
**Describe the bug**
When, for example, I slice a file in multiple parts and turn them objects, cut objects don't keep position and automatically drop on the table.

**3mf File for This Bug**
ANY

**To Reproduce**
1. Open STL
2. Cut in multiple parts
3. Turn them objects
4. Objects fall to the bed! What if I just wanted to remove a section but re-assemble the rest after cutting???

**Expected behavior**
Not dropping the cut objects to the table!

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
MacOS

MadMax389, 2023-04-13:
Split them to parts, not objects.

zhimin-zeng-bambulab, 2023-04-24:
@bolsoncerrado 
First you can cut object to parts.
Then if you want to remove a subpart after cut, you can switch to "objects" as show below, select and delete the subpart.
[image]

bolsoncerrado, 2023-04-24:
WHat I want is able to cut in any amount of slices and manipulate each slice individually BUT the slices staying in the same place they were cut!!

bolsoncerrado, 2023-04-24:
I open any stl.
I slice it.
In order to do any thing with the slices I just created, I must turn them to OBJECTS.
At the very moment I turn the parts into OBJECTS, they "fall" on the bed.

zhimin-zeng-bambulab, 2023-04-24:
I‘m sorry that we currently do not support keeping the position after splitting the model into objects. What exactly do you want to do? Let’s see if there are other ways to achieve the same effect.

bolsoncerrado, 2023-04-24:
Well the only other way is what I said above:

> well found some sort of workaorund like having to switch back and forth from GLOBAL to OBJECTs, delete, save, undo, delete and save again in order for the SAVE AS STL the pieces each time.... what a waste.

bolsoncerrado, 2023-04-24:
In S3D for example you drop the cuts on the table only if you want to.

zhimin-zeng-bambulab, 2023-04-27:
Thank you very much for your suggestion, but due to some restrictions on the current software, it cannot meet your requirements, we will improve it in the future.