bambulab/BambuStudio #1608: Disconnection AMS during printing

DarkDeddok, 2023-04-13:
Disconnection AMS during printing and this is not the first time. After reboot printer, AMS connect to the printer
why is this happening?

[image]

DarkDeddok, 2023-04-13:
red indicators are lit on the AMS

https://user-images.githubusercontent.com/57879696/231823979-ecb2096d-d486-495f-9bfa-84ddbd0c66ae.mp4

christianjou, 2023-04-13:
Might be too much humidity ??

DarkDeddok, 2023-04-14:
not sure
as you can see, I installed additional moisture absorbers

moust4ki, 2023-07-12:
I can also reproduce this issue since the last firmware update (AMS + P1P). 
Its not on every print, kind of random. AMS seems disconnected but continues to print. Its reconnects randomly on the Bambu Studio UI. Sometimes it gives me unknown spool on each bay and after few seconds the spools color and materials reappears.

moust4ki, 2023-07-29:
On my side I can regain AMS control by unplug the power cord. Before losing control to the AMS the state of AMS change randomly like this:

[image]

[image]

[image]