bambulab/BambuStudio #1611: Setting Z-Hop to zero or low values causes slicers bugs when used with color changes

Launert, 2023-04-14:
**Describe the bug**
When you set z-hop in spiral mode to <= 0.0001 or in normal mode to zero, the models do not get sliced correctly any longer.
It does not matter if you set it on the printer settings or if you overwrite it in the filament settings.

**3mf File for This Bug**
https://www.thingiverse.com/thing:763622
[3DBenchy.zip](https://github.com/bambulab/BambuStudio/files/11232802/3DBenchy.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Printer and nozzle settings
2. Click on Extruder Tab
3. Scroll down to "Z-Hop when retracting"
4. Set Value to 0 in normal mode or <= 0.0001 in spiral mode
5. Go to the color tool and paint the chimney in a different color
6. Slice the model

**Expected behavior**
The model should slice normally without layers hanging in the air and parts of the model missing

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 21H2
 - Bambu Studio: 1.5.0.61

DzzD, 2023-04-16:
Until they correct this, if you want a workaround, I have used 0.001 with success.

[image]

SaltWei, 2023-04-17:
Already fixed in internal version. And it will be published soon

SaltWei, 2023-04-18:
v1.6.0 Public Beta has been published

closed 2023-04-18 by SaltWei