bambulab/BambuStudio #1613: Add on-screen/slicer button to perform mid print nozzle wipe

thetoadsage, 2023-04-15:
Sometimes during printing especially PETG the nozzle can get a blob on it that needs to be removed or it's going to get stuck on your print and mess it up. Add a button that will pause the print, purge some filament, perform a few healthy nozzle wipes and then get back to printing. I could have used this many times.

StoneLiBambulab, 2023-04-16:
Hi @thetoadsage ,
Good suggestion. We'll discuss it.

Thanks!
Stone Li

thetoadsage, 2023-04-19:
> @thetoadsage
> There is already be filament purging, nozzle wiping when resuming a printing job. We can optimize the detailed action in resuming in the future firmware release. So when this happens, the user can recover with a pause/resume. Is this satisfying your requirement?

Yes I believe so that sounds like a good solution!

zeekens, 2023-05-09:
I'm looking for this function exactly, just to force a nozzle cleanup mid print

treb42, 2023-07-13:
> @thetoadsage There is already be filament purging, nozzle wiping when resuming a printing job. We can optimize the detailed action in resuming in the future firmware release. So when this happens, the user can recover with a pause/resume. Is this satisfying your requirement?

That nozzle wipe after a pause, for the 6 prints I have used it today, always drags 2-4 cm of material back onto the print which I had to try and get from the nozzle before it could mess up the print.

bani6809, 2023-08-01:
the nozzle wipe after pause is not very good.

here's a better wipe:

G92 E0
G1 E-[retraction_length] F1800
G0 X65 Y245 F12000 ; move to safe pos
G0 Y265 F3000
M400 P100
M971 S11 C10 O0
G92 E0
G1 E[retraction_length] F300
G0 X100 F5000
G0 X70 F15000
G0 X100 F5000
G0 X70 F15000
G0 X100 F5000
G0 Y255

bani6809, 2023-08-01:
#1091 is a duplicate of this

ERSBrad, 2024-11-05:
Please just make this a general button that someone can press before printing as well. Often I load filament or just look at the camera and filament is attached to the nozzle. I always have to click the X, Y buttons a bunch of times to get it to the nozzle cleaner and try and wipe it off. If you do it wrong, filament can fly out onto the build plate when you're printing remotely.