bambulab/BambuStudio #1614: [Feature] Zoom AMS Humidty Indicator

frasier2k, 2023-04-15:
Please show the AMS humdity indicator also in the tool tip window or at least increase the size. There is enouph space. The icon is very hard to view.

[image]

walterwongbbl, 2023-04-17:
Hi.
A new version is about to be released, and we have optimized the humidity icon.

gitX14N, 2023-07-03:
I really, really want this... my eyesight isn't what it once was, and my printer room has an average humidity in the mid 60s. Staying on top of desiccant in the AMS is important. 

I see that this feature was commented as being "about to be released" on Apr 16th. It's July... any chance you can bump this into the next build? Thanks all.

ArmedLeftist, 2023-07-04:
Again, yes, I too would (still) really appreciate that. Then I wouldn’t have to leave my 5x magnifying glasses that I use for tying fishing-flies hanging next to my printer. Thank you!

closed 2023-12-21 by walterwongbbl