bambulab/BambuStudio #1615: Changing filament is not possible when the support interface is selected

DokuKaefer, 2023-04-15:
As soon as you select something in the support interface, it is no longer possible to change the filament above a certain layer height.

[image]

 - OS: Windows 11
 - Bambu Lab Version 1.5.0.61

LiZ-Li-BBL, 2023-04-21:
This is a mistake, we're about to fix it

PixelKaje, 2023-06-16:
Is there an update with this issue? It's still occurring, and prevents me from using any kind of supports on all of my multi-filament prints. Thanks!

LiZ-Li-BBL, 2023-07-21:
This issue has been fixed, please update your version of BambuStudio to this :
https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.65

closed 2023-08-18 by DokuKaefer