bambulab/BambuStudio #1616: Overhang Visualisation and Support Brim

Nosnbora, 2023-04-15:
Hi, 
it would be a huge benefit if Bambu Studio has a preview of red areas of parts of an object which are steeper than the set max angle for support generation like cura does. 

also i really miss the possibility to give my supports, especially tree supports, a support brim for better bed adhesion. 

Thats what i miss the most from cura when i have to slice my custom technical parts.

Best Regards,
Dom

StoneLiBambulab, 2023-04-16:
Hi @Nosnbora ,
    Thank you for your advice, very useful feature,  we will add it after discussion.

Thanks!
Stone Li

Guilouz, 2023-07-08:
@StoneLiBambulab I saw that you added the possibility to Show Overhang in the last beta version, but would it be possible to save this parameter because currently it deactivates with each restart of Bambu Studio.