bambulab/BambuStudio #1627: 1.6 Beta - Height Range Modifier

ECKEROTH, 2023-04-17:
**Describe the bug**
When adding a Height Range Modifier, it default to filament in slot 1

**To Reproduce**
Steps to reproduce the behavior:
1. Make a cube with color in slot 2
2. Add height range modifier to cube
3. New height range color is now based on slot 1 instead of inherit from object in slot 2.

**Expected behavior**
Expected new height modifier object to inherit color from object in slot 2.

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Public Beta 1.6.0

StoneLiBambulab, 2023-04-18:
Hi @ECKEROTH 
Yes, it will be modifed.

Thanks!
Stone Li

LiZ-Li-BBL, 2024-01-16:
Have been optimized. Height range color has a "default" color based on the object.

closed 2024-01-16 by LiZ-Li-BBL