bambulab/BambuStudio #1630: Feature Request: Move from FTPS to SFTP for Improved Security and Reliability

aequasi, 2023-04-17:
Hello BambuStudio maintainers,

First of all, I would like to express my gratitude for your hard work and dedication to the BambuStudio project. Your efforts have been instrumental in streamlining the development process for many users.

I am writing this feature request to propose a switch from the current FTPS (File Transfer Protocol Secure) implementation to SFTP (SSH File Transfer Protocol) for file transfers in BambuStudio. This change would offer several advantages in terms of security, reliability, and ease of use.

## Downfalls of FTPS:

1. Complexity: FTPS uses two separate connections for control and data transfer, which can lead to complications when configuring firewalls and Network Address Translation (NAT) traversal. This makes it more difficult to set up and maintain compared to SFTP, which operates over a single connection.

2. Encryption negotiation: FTPS relies on SSL/TLS for security, but not all servers support the same encryption ciphers and algorithms. This can result in compatibility issues or weaker encryption than desired.

3. Passive and active modes: FTPS requires the use of passive or active mode for data transfers, which can create additional challenges when configuring firewalls.

## Benefits of SFTP:

1. Security: SFTP operates over the Secure Shell (SSH) protocol, which provides robust authentication and encryption. It supports a wide range of encryption algorithms and ensures that both the data and the commands are encrypted, offering a higher level of security.

2. Simplicity: SFTP uses a single connection for both control and data transfer, making it easier to set up and maintain. This also simplifies firewall and NAT configuration, as only one port needs to be opened.

3. Reliability: SFTP includes built-in support for resuming interrupted transfers, ensuring that file transfers are more reliable and less prone to errors.

4. Platform independence: SFTP is widely supported across various operating systems, making it a more versatile option for developers and users alike.

I believe that making the switch from FTPS to SFTP would greatly benefit the BambuStudio project and its users. By providing improved security, reliability, and ease of use, SFTP would help make BambuStudio an even more valuable tool for the development community.

Thank you for considering this feature request, and I look forward to hearing your thoughts on this proposal.

sodachen22, 2023-04-18:
I am sorry, there is no plan to support SSH for the end user. I am not sure if there is solution to support SFTP without SSH.

Hologos, 2023-04-18:
> I am sorry, there is no plan to support SSH for the end user. I am not sure if there is solution to support SFTP without SSH. Do you have any suggestion on SFTP without SSH?

OP doesn't want you to implement SSH for end-users. He just asks you to replace FTPS protocol with SFTP which internally uses SSH. :-)

aequasi, 2023-04-18:
You can set up SFTP (and SSH) to make sure the user only has access to SFTP

jrsphoto, 2023-04-18:
correct, you need SSH for SFTP, but as aequasi mentions you just edit the sshd_config file to only allow sftp.  No end-user access to SSH at all is needed.

sodachen22, 2023-04-23:
Yes, SFTP and FTPS can be used in different scenarios. There is an article describing the difference https://www.spiceworks.com/tech/networking/articles/sftp-vs-ftps/. 
In the original design, we wanted the protocol to have better performance than better firewall compatibility
1. In our testing, FTPS is about 30% faster than SFTP for transferring a large 3MF file in the same LAN environment.
2. SFTP commands offer greater control than FTPS commands, such as chown, chmod, ln or symlink. They are not needed for a file tranmission. 
So FTPS is used here and here is also an article about the protocol and prots to the printer: https://wiki.bambulab.com/en/general/printer-network-ports. Is there any difficulties in using FTPS in some network environments or any stability issues?

Majestic7979, 2023-04-25:
Ah... "security" again. Who's going to 1. find your printer IP on the LAN, 2. find your LAN code on the printer screen and 3. access your files? I mean it's overkill. Nobody will be able to login to the ftp server without the lan code which is the password. That's your security. Not sure why people get crazy about things like these nowadays. It's geometry files and timelapses not banking information or amateur p*rn stored in the memory. We need many new improvements and they don't have unlimited workforce, so hopefully they will focus on what we need before thinking about changing what is not broken.

aequasi, 2023-04-25:
@sodachen22 

1. The bigger problem is inconsistency. While yes it may be faster, SFTP is significantly more reliable. Even HTTP is more reliable.
2. Sure, I agree. Control wasn't something we mentioned in the feature request. Its not something I'm using as a "selling point" to switch to SFTP

---

While I appreciate your input @Majestic7979 , it's a bit shortsighted. 

Just because you don't have a valid need for security doesn't mean others don't. Just because you don't see value in a feature request doesn't mean others can't.

Besides that, there were more points listed than just security.

Majestic7979, 2023-04-25:
> @sodachen22
>
> 1. The bigger problem is inconsistency. While yes it may be faster, SFTP is significantly more reliable. Even HTTP is more reliable.
> 2. Sure, I agree. Control wasn't something we [...]

It's all related to security. Firewall = security. Bambu Lab manages the set up and maintenance. This has nothing to do with the user. The FTP in the printer is user-accessible but it was never advertised as a feature. Bambu Lab responded above as to why they went this route and I agree with them that speed is more important than some people's paranoia. I'm not saying it's invalid, it's simply paranoia in this case. In terms of resuming transfers, I don't see why they should re-architect their firmware and waste their development time on this when the current solution already works and the user can just move the printer closer to the router or use an extender. If we were transferring several GBs of data at once I could see the benefit but for <100MB it's completely a waste of time to re-develop the solution.

Be realistic, this is not needed and this whole security argument is only a way to generate guilt and force the developer to change things that don't need changing. There is zero security implications in this use case. If someone can access the FTP it's because they physically have access to the printer, then the person has bigger problems to deal with that means door access and that's not Bambu Lab's issue. I am not blindly defending BL, I am being realistic.

Hologos, 2023-04-25:
First of all, we want to transfers gigabytes of data - timelapses. The other this is - it's not paranoia. I don't want to be rude but with this sentence `Who's going to 1. find your printer IP on the LAN`, you have proven you don't know how devices get compromised nowadays. It's not a person who scans a network, it's bots. And they will find the device with a single broadcast packet.

The main point is that if they implement things such ways, how can we be sure there are no security holes when the firmware is closed source.

Also, nowadays teams rarely reimplement things that have been implemented several times, they would use a library and instead of passing instance of one object, they pass the other. And if they use proper developer conventions, they just make a wrapper class so the API would not change. 🤷‍♂️

Majestic7979, 2023-04-25:
> First of all, we want to transfers gigabytes of data - timelapses. The other this is - it's not paranoia. I don't want to be rude but with this sentence `Who's going to 1. find your printer IP on [...]

Oh God. If a bot finds a printer on the LAN, I'm sure your bank account will be compromised. Facepalm. What will they do, post your 3D-printed dildo timelapses in the dark web and demand that you pay ransom? I am being sarcastic, no offense intended. I mean come on that's pure paranoia. This is not a security hole. Not everything has to be encrypted. The right tools for the job comes to mind. It's like saying you need to hire a crane to lift a burger from the plate on your table when you can just use your hands.

As for transferring timelapses, Ok I will agree that this is a very good example. But then use that as an example. There is absolutely no "security hole". Anyone talking about security in this context is simply being paranoid and denying this just makes the person look silly. It's a 3D printer, not a bank account or a baby monitor/security camera.

aequasi, 2023-04-25:
@sodachen22 Please lock this conversation. Its quickly going to turn uncivil