bambulab/BambuStudio #1634: [v1.6.0 Public Beta BUG] - No more space on SD Card

Obicom, 2023-04-18:
**Describe the bug**
I tried the latest Bambu Studio Public Beta (1.6.0.65) and if I try to send a file to the printer (send to SD card) I get the error message (see attachment) "not enough space on SD card. It is a 512GB SD card and more then 400GB are free. I deleted a lot of files and videos but still the same error message.

**Expected behavior**
I expect that Bambu Studio is able to send a plate to the scard without an error if the sd card has more then 400 GB free space.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

walterwongbbl, 2023-04-21:
Hi. 
Please provide the Studio LOG file when the error occurred.
I will analyze this issue.

walterwongbbl, 2023-05-26:
Hi.
We found that the error prompt in Studio is not accurate and we optimized the description of printing errors.
This will be greatly improved in the next version

closed 2023-10-13 by walterwongbbl