bambulab/BambuStudio #1635: [v1.6.0 Public Beta BUG] - Last button is cut in Filament/AMS view

Obicom, 2023-04-18:
**Describe the bug**
Last button is cut in Filament/AMS view (German Language)

**Expected behavior**
All buttons should be full visible

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

Guilouz, 2023-04-19:
How do you obtain Ext Spool, my X1C is updated to latest version and latest Bambu Studio too and I don't have it (tested on macOS and Windows).

walterwongbbl, 2023-04-20:
> How do you obtain Ext Spool? My X1C is updated to latest version and latest Bambu Studio too and I don't have it (tested on macOS and Windows).

[image]
[image]

Hi.
Please check your config files and uninstall the old Bambu Studio before installation

Guilouz, 2023-04-20:
@walterwongbbl Yes it's enabled for my X1C : 

[image]

But External spool doesn't appear. I have tested on macOS et on Windows.

Guilouz, 2023-04-20:
@walterwongbbl I have tried many things, with or without AMS, complete uninstallation of BambuStudio and reinstall without reloading my cloud configuration and same issue, ext spool doesn't appear.

walterwongbbl, 2023-04-21:
Hi.
Please check the printer screen.  Is there an ext spool present.On the control page of AMS

Guilouz, 2023-04-21:
@walterwongbbl No i don’t have it and my bambulab is up to date.

walterwongbbl, 2023-05-11:
Hi.
Does this issue still exist

Obicom, 2023-05-11:
> Hi. Does this issue still exist

It is still the same ... (1.6.0 Beta 2)
[image]

walterwongbbl, 2023-05-18:
Hi.
We will fix the UI display issues in the German language environment.
This fix will be included in the next version.

Guilouz, 2023-05-18:
@walterwongbbl This happens with french language too.

walterwongbbl, 2023-10-13:
Please download and use the latest version of Studio

closed 2023-10-13 by walterwongbbl