bambulab/BambuStudio #1636: [v1.6.0 Public Beta BUG] Limited AMS color picker

Hologos, 2023-04-18:
**Describe the bug**
The AMS color picker on the Device tab is now limited in color selection. The color picker in filament section is still fine. I have a lot of different gray filament colors and now I can distinguish between them.

**3mf File for This Bug**
not relevant

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Device tab.
2. Click on AMS slot.
3. Click the color button.
4. See limited color picker.

**Expected behavior**
Previous (system) color picker was way better IMHO.

**Screenshots**
AMS color picker
[image]
Filament section color picker
[image]

**Desktop (please complete the following information):**
 - OS: macOS 13.3.1
 - Version Bambu Studio v1.6.0.65

sodachen22, 2023-04-23:
1. AMS color can be set with Studio/Handy/Screen of the printer, the screen of the printer can not set as much color as the Studio.
2. AMS filament backup feature support the backup filament in exactly the same color, which means AMS can switch to another filament with exact the same properties when filament running out, the properties includes color.
3.  1 and 2 require the color set with Studio/Handy/Screen of the printer to have exactly the same RGB value, so we make Studio/Handy/Screen of the printer have the same color table with the same RGB value.

Hologos, 2023-04-23:
But now I can't have loaded different shades of colors because if one runs out, it'll use a different shade. I'd have to disable the filament backup feature. And I'd have to turn it back on if my spool has not enough filament to finish the print.

Would it be possible to make this an option in settings? I understand for new users it's easier but for morr advanced users, this is really bad. And for p1p users without LCD, this is not even necessary.

Majestic7979, 2023-04-25:
> 1. AMS color can be set with Studio/Handy/Screen of the printer, the screen of the printer can not set as much color as the Studio.
> 2. AMS filament backup feature support the backup filament in [...]

Then offer a way to link slots instead! You are arbitrarily removing features that people are used to instead of putting on your thinking hat and thinking outside the box. Simply allow the user to "link" slots so that prints can continue when the first slot is empty. You could add a checkbox under each slot to activate failover. Or a button to open a matrix, where the user can check a box to link AMS unit A Slot 1 to AMS unit B slot 3 which both contain white PLA for example. This all without relying on locking down the editing or RGB colors. Please STOP locking things down/removing features. There is always a 2nd alternative that works better than removing things that users have got used to.

sodachen22, 2023-04-25:
Yes, an user defined linked slot is a solution we are considering. Thanks for the feedback.

Majestic7979, 2023-04-25:
> Yes, an user defined linked slot is a solution we are considering. Thanks for the feedback.

Great. I look forward to seeing RGB editing return and a better linking solution implemented.

Hologos, 2023-05-03:
@sodachen22 I downloaded the latest beta and I still can't set a custom color. Clicking on custom color doesn't do anything and selecting pre-defined color is super slow on macOS (it takes 5 seconds to change the color).

[image]

walterwongbbl, 2023-05-04:
> @sodachen22 I downloaded the latest beta and I still can't set a custom color. Clicking on custom color doesn't do anything and selecting pre-defined color is super slow on macOS (it takes 5 seconds to change the color).
> 
> [image]

Hi.
We tested using the 13.3.1 system and it appears to be normal.

Hologos, 2023-05-04:
Hello, see attached gif how it behaves on my machine. I included mouse click highlight. Some colors I can choose, but others I can't. Also, I am not able to open system color picker. I managed to open it twice with some kind of swipe move but I am not able to replicate it anymore.

If you need me to provide you with some logs, let me know.

[image]

walterwongbbl, 2023-05-06:
Hi.
Thank you very much for providing the gif.
I saw the issue you described. We tried to reproduce this problem on a MacBook Pro but were unsuccessful. I suspect it may be related to the mouse click events on macOS, and I will try to fix this issue.

sam43434, 2023-05-29:
> @walterwongbbl Hello, see attached gif how it behaves on my machine. I included mouse click highlight. Some colors I can choose, but others I can't. Also, I am not able to open system color picker. I managed to open it twice with some kind of swipe move but I am not able to replicate it anymore.
> 
> If you need me to provide you with some logs, let me know.
> 
> [image] [ [image] ](https://user-images.githubusercontent.com/1620004/236134491-05c357a6-f37d-4063-9381-b04c8e4eee0a.gif) [ ](https://user-images.githubusercontent.com/1620004/236134491-05c357a6-f37d-4063-9381-b04c8e4eee0a.gif)

Same issue here on MacOS 13.4,   Having to reset the button the filament and open and close the filament window before it will let me select a colour

walterwongbbl, 2023-05-30:
> > 您好，请参阅随附的 gif 它在我的机器上的行为。我包括鼠标单击突出显示。有些颜色我可以选择，但有些颜色我不能。另外，我无法打开系统颜色选择器。我设法通过某种滑动移动打开了它两次，但我无法再复制它了。
> > 如果您需要我为您提供一些日志，请告诉我。
> > [image]
> >       
> >         [
> >           
> >         
> >             [image]
> >           ](https://user-images.githubusercontent.com/1620004/236134491-05c357a6-f37d-4063-9381-b04c8e4eee0a.gif)
> >         
> >         
> >           
> >             
> >               
> >             
> >             
> >               
> >               
> >             
> >           
> >           [
> >             
> >               
> >             
> >           ](https://user-images.githubusercontent.com/1620004/236134491-05c357a6-f37d-4063-9381-b04c8e4eee0a.gif)
> >         
> >        [ [image] ](https://user-images.githubusercontent.com/1620004/236134491-05c357a6-f37d-4063-9381-b04c8e4eee0a.gif)
> >       
> >         [
> >           
> >         
> >             [image]
> >           ](https://user-images.githubusercontent.com/1620004/236134491-05c357a6-f37d-4063-9381-b04c8e4eee0a.gif)
> >         
> >         
> >           
> >             
> >               
> >             
> >             
> >               
> >               
> >             
> >           
> >           [
> >             
> >               
> >             
> >           ](https://user-images.githubusercontent.com/1620004/236134491-05c357a6-f37d-4063-9381-b04c8e4eee0a.gif)
> >         
> >        [ ](https://user-images.githubusercontent.com/1620004/236134491-05c357a6-f37d-4063-9381-b04c8e4eee0a.gif)
> 
> 同样的问题在MacOS 13.4，Bambu Studio Beta 1.6.2.4上
> 
> 必须使用灯丝窗口上的重置按钮并打开和关闭灯丝窗口，然后才能让我选择颜色

Hi.
This is an issue caused by the characteristics of the Macos 13. x system.   
We tried to avoid this problem. Will be included in the next version.

Hologos, 2023-06-26:
@StoneLiBambulab @sodachen22 @walterwongbbl Any progress on this? I can't use my printer with AMS now. You made my setup a brick with an update.

Hologos, 2023-07-07:
Fixed, thank you! 👏

closed 2023-07-09 by sodachen22