bambulab/BambuStudio #1637: [v1.6.0 Public Beta BUG] Filament oozing from nozzle in the beginning of the print

Hologos, 2023-04-18:
**Describe the bug**
`start_gcode` is now changed to preheat the nozzle to initial temperature. It makes filament ooze (even PLA) from the nozzle. It gets thrown on a random place on the heatbed when it tries to clean the nozzle. Since it's too long, it doesn't get to the shoot.

```diff
diff --git a/old.txt b/new.txt
index 8462681..c734243 100644
--- a/old.txt
+++ b/new.txt
@@ -1,5 +1,7 @@
 ;===== machine: P1P ========================
-;===== date: 202230129 =====================
+;===== date: 20230323 =====================
+;===== turn on the HB fan =================
+M104 S[nozzle_temperature_initial_layer] ;set extruder temp to turn on the HB fan
 ;===== reset machine status =================
 G91
 M17 Z0.4 ; lower the z-motor current
```

**3mf File for This Bug**
not related

**To Reproduce**
Steps to reproduce the behavior:
1. Slice a model with new profiles.
2. Send it to printer.
3. Observe the behavior.

**Expected behavior**
No oozing at the start.

**Screenshots**
I'll post a video once my print gets finished.

**Desktop (please complete the following information):**
 - OS: macOS 13.3.1
 - BambuStudio: v1.6.0.65

MadMax389, 2023-04-19:
I have noticed this too, I was wondering what was going on. I suppose we can just comment out that line?

Hologos, 2023-04-19:
> I have noticed this too, I was wondering what was going on. I confirmed this added line is in the X1C profile as well, and inherited by my custom profiles. I suppose we can just comment out that line?

I think they added that to speed the preparation phase but this is unwanted side-effect.

Majestic7979, 2023-04-25:
Ahhhhh so this is why my build plate is messier now that I'm running the beta. Yep definitely not wanted.

QingZhangBambu, 2023-04-25:
We will fix it in next release. For now, you can use "M104 S60" to avoid it

Majestic7979, 2023-04-25:
> We will fix it in next release. For now, you can use "M104 S60" to avoid it

When you say "next release" do you mean the official one or the beta 1.6 beta?

QingZhangBambu, 2023-04-25:
official 1.6 version or 1.6 hotfix

Majestic7979, 2023-04-25:
Thanks, great.

closed 2023-06-13 by QingZhangBambu