bambulab/BambuStudio #1640: Bambu Studio fails to run with Fedora 38

A-nglea, 2023-04-19:
Both app images 1.05 and 1.06 fail to run on the new version of Fedora namely 38. (Fedora updates at 6 monthly intervals)

Two library files raised "not found" errors:
libwebkit2gtk-4.0.so.37, and 
libjavascriptcoregtk-4.0.so.18 

Those were easily remedied by creating symbolic links pointing to:-
libwebkit2gtk-4.1.so.0      (which itself points to    libwebkit2gtk-4.1.so.0.8.1), and
libjavascriptcoregtk-4.1.so.0      (which itself points to  libjavascriptcoregtk-4.1.so.0.3.9)

Then when run Bambu studio failed to start with an error
(process:4019): libsoup-ERROR **: 11:19:20.788: libsoup2 symbols detected. Using libsoup2 and libsoup3 in the same process is not supported.

Interestingly both the Prusa Slicer and Cura run on both version 37 and 38 of Fedora.

alextrical, 2023-04-20:
Same issues here, I can confirm that the appimage's built here have the same issues on Fedora 38 for me.
The Prusa Slicer, and Superslicer app images load without issue

lanewei120, 2023-04-21:
how about using libsoup2.4?
the image is built from Fedora 36,  currently we don't know how to build a image universal for different os versions
we will try to solve this in the future

currently, coult you try to use libsoup-2.4?

A-nglea, 2023-04-21:
Thank you for your refreshing  frankness and honesty! It is much appreciated. 

Libsoup-2-4 is not available for the current Fedora version.

I've tried to compile from source but all references apply to Ubuntu systems (dpkg and apt especially) Fedora's dnf or yum do not use a similar syntax. I tried translating and got as far as ./BuildLinux.sh -r running and reporting missing packages that are in fact installed. The differences between dpkg and dnf overwhelmed my ability.

I made a search of the git clone for Bambu Studio and could only find reference to libsoup2.4-dev in Containerfile and Dockerfile.  If you were to change those libsoup2.4 references to libsoup3-dev and recompile I imagine it  then may work - libsoup3-dev should be backwards compatible.

lanewei120, 2023-04-24:
I am not sure it will work or not when switching to libsoup3 on Fedora 36 bwith libsoup2.4 installed(Previously some customer reported issue with Fedora 36)

do you have a image built by yourself using libsoup3?
We can also have a try on the fedora36 machine here

A-nglea, 2023-04-24:
[image]
Building dependencies fails at 47% (Tried from scratch twice with same result)

Build instructions are confused with references to BambuStudio_dep
as in 

> create a directory to store the dependence built: /home/username/work/projects/**BambuStudio_dep**
(Please make sure to replace the username with the one on your computer)

cd BambuStudio/deps
mkdir build;cd build

cmake ../ -DDESTDIR="/home/_username_/work/projects/**BambuStudio_dep**" -DCMAKE_BUILD_TYPE=Release 

and in 3. building the Bambu Studio

> cmake .. -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="/home/_username_/work/projects/**BambuStudio_dep/usr/local**" -DCMAKE_INSTALL_PREFIX="../install_dir" -DCMAKE_BUILD_TYPE=Release

I see other users are having trouble with Ubuntu too now.  I have a virtual Ubuntu version 22.04 running Bambu Studio. But when I first created a virtual Ubuntu machine, I allowed it to update as it installed. Bambu Studio failed to run on that machine and complained with the same error in my first post. 
Ubuntu was deleted and a new virtual instance was created from an .iso, without allowing updates, and now Bambu Studio runs fine in Ubuntu 22.04.

spuiuk, 2023-04-24:
I was trying to run the latest Bambulab provided appimage from
https://github.com/bambulab/BambuStudio/releases on a fresh install of Fedora 38. 

I had to install the following packages to get it to work.
webkit2gtk4.0  mesa-libOSMesa javascriptcoregtk4.0

There are a few more javascript errors and my live stream doesn't work but I can send sliced prints over the cloud.

I hope this helps.

A-nglea, 2023-04-24:
> I was trying to run the latest Bambulab provided appimage from https://github.com/bambulab/BambuStudio/releases on a fresh install of Fedora 38.
>
> I had to install the following packages to get it to [...]

Thanks  v. helpful. Got all those packages. So it seems like updating from 37 to 38 has causes inconsistencies. Not sure I can face doing a complete re-install :(

closed 2023-04-24 by A-nglea

spuiuk, 2023-04-25:
The older version of webkit2gtk uses calls to the old libsoup2 version and the new version of the package uses libsoup3. Since your symlinked libraries uses the newer libsoup3 and the app has been built by libsoup2, there is a mismatch which you see. 
The webkit2gtk4.0 package is the older version with libsoup2 provided for compatibility reasons and therefore works with the  Bambulab appimage.

A-nglea, 2023-04-25:
Thanks for clarifying, that's really helpful.