bambulab/BambuStudio #1641: [v1.6.0 Public Beta BUG] Login in screen is blank....nowhere to enter credentials

managerman, 2023-04-19:
When trying to login, a blank login screen is shown with no place to enter credentials

Steps to reproduce the behavior:
1. Click on login/Register
2. Blank login screen shows

[image]

**Desktop (please complete the following information):**
 - OS: [Windows 11]
 - Version 22H2 22624.1610

lanewei120, 2023-04-24:
sorry for the late reply
we just missed this issue

it should be a network related issue, 
is this issue still there?

lanewei120, 2023-08-08:
should be network related issue
close it currently
if still has this issue, let's reopen it

closed 2023-08-08 by lanewei120