bambulab/BambuStudio #1645: AMS warning message not confirmable

shompy12, 2023-04-20:
I updated to the public Beta of Bambu Studio 1.6 and now I get this warning message that it is not recommended to print with CF materials in the AMS. I cannot confirm this message no matter how often I click the confirm button. It will just close the little window and stay idle on the send print job to printer window below it. The software will not let me print with this filament. It will not warn me for printing with Bambu PETG-CF or PA-CF which is very odd. Printing with the PET-CF under 1.5 was never a problem for me.

Steps to reproduce the behavior:
1. Load PET-CF filament in AMS
2. Slice model with PET-CF filament
3. Print Plates
4. Click on send
5. Message appears and stays in loop

**Expected behavior**
I expect to be warned and then I should be able to print. How else should I do prints with CF and support amterials?

**Screenshots**
[image]

**Desktop (please complete the following information):**
Windows 10
Bambu Studio 1.6.0.65

walterwongbbl, 2023-04-20:
Hi. 
Are the filaments you use from Bambu brand？

shompy12, 2023-04-20:
Hey,

yes they are. In my AMS I have all Bambulab PETG-CF, PET-CF, PAHT-CF and PLA.

walterwongbbl, 2023-04-20:
Hi.
[image]
AMS does not support bambu's pet CF filament.  But if you really want to use it for printing,
Do not let AMS recognize the RFID identification of the filament.

shompy12, 2023-04-20:
Yes, I am aware of that. The AMS also is not compatible with PAHT-CF yet the error does not occur. The next question would be: Why even bother with an RFID chip then? And how to print with support materials? I understand the error message. I am fully fine with telling the printer "Yes, I know it wears down the AMS" but please let me print with the materials I choose to load in the ams. This should not grow into a discussion about AMS wear but more about "why the heck can I not confirm this warning?"

shompy12, 2023-04-20:
Okay I understand your plan to protect novice users. Nonetheless I would be happy to see that there was an option to discard the message or to have a check box in the filament settings to override this warning to not artificially limit those users that are aware of the consequences. The filament is even advertised as "engineering grade" so a novice user should probably not use it but that is less fact but more my opinion.

closed 2023-04-20 by shompy12

walterwongbbl, 2023-04-20:
> Okay I understand your plan to protect novice users. Nonetheless I would be happy to see that there was an option to discard the message or to have a check box in the filament settings to override [...]

Hi.
Thank you for your suggestion. We will add this option in the next version (in developer mode).