bambulab/BambuStudio #1647: Advanced Connector Cut - Rotating An Object Before Cutting Does Not Rotate The Hole Negative Volume (BS 1.6 PB)

MadMax389, 2023-04-20:
**Describe the bug**
Rotating an object before placing the connector dowel does not rotate the hole for the dowel.

**To Reproduce**
Steps to reproduce the behavior:
1. Place a cylinder on the plate and rotate it 90 degrees
2. Place a dowel on the cut surface
3. Perform cut and notice the negative volume for the dowel did not rotate with the object

**Expected behavior**
Hole should conform to the cut surface

**Screenshots**
Place cylinder and rotate it 90 degrees
[image]

Place a dowel on the cut surface
[image]

Notice the hole for the dowel is horizontal instead of vertical
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Win 11
 - Bambu Studio 1.6 Public Beta

zhimin-zeng-bambulab, 2023-04-27:
It will be fixed in the next version.

zhimin-zeng-bambulab, 2023-08-18:
You can have a tay in the latest version

closed 2023-08-18 by zhimin-zeng-bambulab