bambulab/BambuStudio #1648: Bambu Studio crashes hard when using camera view and RTX 3070 Ti Laptop GPU

sprior, 2023-04-21:
**Describe the bug**
I have a Lenovo Thinkpad P1 Gen 5 with RTX 3070 Ti Laptop GPU and 4K LCD display and Win 11.  When I run Bambu Studio (all versions since last December when I got my Bambu) with the lower power Intel GPU I can go to the Device tab and start the camera view, but when working with more complex models the performance starts to suffer.  However when I go into System -> Display -> Graphics and create a custom options for apps profile for Bambu Studio and confirure a graphics preference to High Performance NVIDIA GeForce RTX 3070 Ti GPU, if I then start BS the Prepare screen is much faster as expected, but if I go to the Device Page and try to start the camera view the program hangs and must be killed to recover.  Going back and removing the custom preference then allows the camera view to operate normally again but with reduced performance in the Prepare screen.

**3mf File for This Bug**
No model required, just the Device camera view.

**To Reproduce**
Steps to reproduce the behavior:
1. On a WIndows 11 laptop  with dual Intel and RTX 3070 Ti graphics
2. go into System -> Display -> Graphics and create a custom options for apps profile for Bambu Studio and configure a graphics preference to High Performance NVIDIA GeForce RTX 3070 Ti GPU
3. Start Bambu Studio, go to the Device screen and start the camera view
4. Bambu Studio crashes hard.

**Expected behavior**
Bambu Studio should be able to run with a high end GPU

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Lenovo P1 Gen 5 laptop with NVIDIA GeForce RTX 3070 Ti GPU
 - Windows 11
 - NVIDIA Game Ready graphics driver

bambu123, 2023-04-21:
Sorry! This is a known issue relate to the old d3d9 driver and we can't fix it until the refactored version.

sprior, 2023-04-22:
That's actually a better answer than I expected - I figured this was going to be a total surprise with limited ability to reproduce elsewhere.  Can you give any clue for a timeline on this refactored version?

bambu123, 2023-04-23:
Sorry, there is no clear timeline now.