bambulab/BambuStudio #1651: Load Profiles from Git Repository

magic-happenz, 2023-04-21:
**Is your feature request related to a problem? Please describe.**
I used to store my slicer profiles in a Git repository to keep track of changes and easily know when I made changes, why and to roll them back. All advantages from a version history apply here.

**Describe the solution you'd like**
Slic3r/PrusaSlicer/Superslicer use to have a parameter "-datadir" that allows to open the application with an explicit path from where it should load/save the profiles. This could be stored in a soft link for example to start the application. That way when changes are made, they are tracked via git and can be committed to a repository. However, this does not work with Bambu Studio.

**Describe alternatives you've considered**
An alternative could be that you set a setting where profiles are stored by the application. 

**Additional context**
Issue is for example explained here: https://github.com/prusa3d/PrusaSlicer/issues/5023

lanewei120, 2023-04-23:
as the user profile will be uploaded to cloud if you select "Auto Sync..."
[image]

it is hard to store it as a local git in the meantime
do you mean use it in the offline mode?

magic-happenz, 2023-04-23:
Don't get me wrong. It is great the Slicer got a backup and Auto Sync feature but I am used to Hit. That it works online and offline is just another advantage but I am just looking for advantages that come with a version history. I like to document the changes I made, especially why. Nothing is more frustrating to realize that you changed a setting in the past that negatively influence your prints and you don't know when, what and why you changed it.

Hologos, 2023-04-24:
I am version controlling my configs from the default directory (on macOS `~/Library/Application Support/BambuStudio`), it causes no problems. Is it suitable to you or do you need a dedicated directory?

magic-happenz, 2023-04-24:
I haven't yet checked where my profiles on Windows are stored exactly but it could be a workaround. I would just have to create a new repo rather than using my existing one for 3D printer profiles.

lanewei120, 2023-04-25:
[image]
you can find the configuration directory here

as there are lots of files in the datadir, change it from command line may cause issues
you can init your git repository from there directly

magic-happenz, 2023-04-25:
I found that folder too, I was just confused because it was empty but makes sense cause I don't have own presets yet. However, as I mentioned I would have to initialize a new Git repo from here, with SuperSlicer for instance I can just point it to an existing one and even reuse profiles among the slicers. Anyways, would be cool if that would be possible but the workaround is not too bad.

magic-happenz, 2023-04-25:
Ok it won't really work out with initializing the repo either I noticed... If you do so you are not really able to commit a whole profile but just the diff between your custom profile and whatever preset you have "copied" from. Means the repository would not reflect the whole profile but just the diffs from the original which is not ideal. Furthermore if you make a change on the profiles lets say in another slicer or just because you edit the JSON and it is by chance the default of the original profile, Bambu studio removes the whole setting from the file which is odd.

Hologos, 2023-04-26:
There is a folder called system with system profiles. So you can track all the changes.

magic-happenz, 2023-05-01:
@Hologos you get me wrong. I know that there is another folder where the system profiles are stored but tracking it won't help me. First I would have to include a lot of noise files in my repo that I do not need there, second I still won't be able to fully track and represent a single profile since as soon as I change a setting that is by coincidence similar to the system one, that setting is removed from my profile. Its not practical to put this in a version control.

Hologos, 2023-05-01:
> @Hologos you get me wrong. I know that there is another folder where the system profiles are stored but tracking it won't help me. First I would have to include a lot of noise files in my repo that [...]

Sorry but I don't follow. With a simple script you can view all parameters of the profile.

Also if you don't want to track "noisy files", you can just ignore them using `.gitignore`.

Also if you change a parameter that is different from the system one, it gets recored into your profile. That is how Bambu designed it, it records only the difference from a profile you inherit the profile from.

magic-happenz, 2023-05-01:
> Sorry but I don't follow. With a simple script you can view all parameters of the profile.

I don't know what do you mean by that.

> Also if you don't want to track "noisy files", you can just ignore them using `.gitignore`.

Yes I know that but it does not solve my initial request.

> Also if you change a parameter that is different from the system one, it gets recored into your profile. That is how Bambu designed it, it records only the difference from a profile you inherit the profile from.

I know how it works, I am requesting to change this like its working with Super Slicer for example. That is what my feature request is about.

Hologos, 2023-05-01:
Sorry, I don't see the issue. As I said, I am managing my config files with git without any problems. I can see every change I've made to my profile (and optionally every change Bambu did to system profiles). Let's stop chatting here, if you'd like, we can get in touch via Discord and talk about it. Cheers :)