bambulab/BambuStudio #1652: Feature Request - paint in non-specific filament

ljm42, 2023-04-22:
Some objects have interior walls which won't be seen once the final model is assembled. A nice optimization would be to paint those walls with "any filament" rather than a specific one, and let the slicer re-use whatever color is already loaded like it does for supports. This will help reduce filament waste from switching colors.

Majestic7979, 2023-04-25:
How will it reduce waste? Genuine question. Doesn't the printer use the same filament for infill, and only changes when it's going to show external stuff? Since you're doing multicolor prints, maybe you can try setting the print order to do infill/internal/outer and that way it's only going to change filament for outer surfaces? Or am I missing something really obvious here?

ljm42, 2023-05-16:
> How will it reduce waste?

On some objects, there are paintable walls where the color really doesn't matter because when everything is assembled the wall will be hidden. This means unnecessary filament switching will happen on some layers. We can reduce waste by letting the slicer use whatever filament is already loaded rather than forcing it to switch colors when it really doesn't matter.

Majestic7979, 2023-05-17:
I see. Thanks.