bambulab/BambuStudio #1655: Ability to choose which horizontal surfaces to be ironed.

smford, 2023-04-22:
**Is your feature request related to a problem? Please describe.**
I would like to choose which surfaces to iron.  Currently BambuStudio only has a limited drop done selection of ironing choices, however there are times when I just want a single surface ironed.  This is important to me, because otherwise I need to choose "Iron all top surfaces" which greatly increases the print time as it now needs to unnecessarily iron much more surface area.

**Describe the solution you'd like**
I would like the ability choose which surfaces by painting that should be ironed. 

Rather than using a paint brush, I would like a feature that just allows me to choose a particular horizontal surface area and choose "iron this".   I envisage this feature to be in the preparation tab of BambuStudio as button that can be selected, and act similarly to the "Lay to face" option where it highlights surfaces that are horizontal and suitable for ironing.

**Describe alternatives you've considered**
I do not know of any other alternatives to consider.

**Additional context**
None.

coat64, 2023-04-23:
Maybe not what you are looking for, but adding a modifier at the right place and set ironing works. You can even iron a part of the surface. 
[image]

[image]

DzzD, 2023-04-23:
Never thought of using modifiers for ironing, great tips thanks

smford, 2023-04-23:
> Maybe not what you are looking for, but adding a modifier at the right place and set ironing for the modifier works. You can even iron a part of the surface.

This is very helpful, thank you for pointing it out, I will try and work with this until my feature request is hopefully accepted.  As some of my designs are complex, using this type of modifier will be challenging on them.  Hopefully my feature request will get developed one day.