bambulab/BambuStudio #1661: Please improve the T g-code speed

DzzD, 2023-04-23:
Making multi-material print is one of the greatest feature of the Bambulab printer, unfortunately color change take ages. 

I have worked on the gcode to change color and achive to get a really good speed boost ([MR requested](https://github.com/bambulab/BambuStudio/pull/1660))

**Unfortunatly there seems to be no controle on the T gcode wich :**
- cut the filament
- go to purge bucket location
- unload current filament
- load new filament
- some little pauses

**I think there some ways to improve it a little :**
- cut a bit faster, currently this take way too long, I understand this is have to be done carefully, but really it can cut at least three times faster without risk.
- use travel speed to go to purge bucket location, for now it is not using the maximum speed
- remove pauses that are not needed

For now, it take 10 seconds before the old filament begin to be rewind, I am pretty sure you can reduce it to 5 seconds.

Using a better filament change gcode and shorters PTFE, I already have achieved to divide by twice the print time of multi-material print.

Two examples using the exact same print/filament/spools and purge volume :
original 43 minutes => modified 21 minutes
original 4h40 => modified 3h00

**Improving T g-code would give even more speed gain.** 

Even 5 seconds multiplied by 100 filament changes represent for the full print 8 minutes !

HobinBambu, 2023-04-26:
Hi Dzzd:
    Thank you for your good suggestion,we are also doing some optimization to reduce the time of switching filament,but the switching reliability for all types and all brands of filaments is a very important constraint.Setting some parameters in the T command will be supported gradually .
1. change the travel speed of "T" command:
You can modify the "F[travel_speed]” in "M620.1 X[travel_point_1_x] Y[travel_point_1_y] F[travel_speed] P[n]"to set the travel speed of "T" commond as below.
[image]
this G-code is already included in studio, and the corresponding X1 firmware will be released recently.The old speed in the T command is 250mm/s, which will be adjusted to 350mm/s(F21000) in this version,and it may be replaced with the user set ”travel_speed“ in studio in the future.
2. change the cut speed:
Will not be supported for now due to reliability.
3. "remove pauses that are not needed"
Do you mean the pause between the tool head moving to the chute and the filament being pulled out?

Thanks.

DzzD, 2023-04-26:
Sure, I understand there are some constraints due to multiplicity of filament, some "placeholders" maybe helpfull to customize some parts depending on filament, please have a look to the PR I made (https://github.com/bambulab/BambuStudio/pull/1660), some people are testing it with succees and it is a lot faster, but surely still not perfect.

1 - Thanks for this tips ! I will try it.

2 - I understand for the cut speed, but it is damnly slow :p, what happen if I begin the cut myself using g-code ?

3 - about pauses, the main one I see is once the new spool finish loading the following g-code after T is not executed immediatly, the purge start few seconds after the new spool is finished loaded. Also note that it take 10 seconds before old filament begin to unload.

Here is a timing using the g-code I proposed (in my PR) : 

**Fulshing volume : 400mm³**

- Filament change start : 0.0s / 0.0s
- Filament cutting (T gcode) : **10.3s** / 10.3s => _this is due to cutting, and a little pause before starting unload_ 
- Filament unloading (spool rotating) : **5.67s** / 15.97s => _this is only due to length of PTFE and spool load speed_
- Filament loading (spool rotating): **9.26s** / 25.23s => _this is only due to length of PTFE and spool load speed_
- Purging the filament into bucket: **16.65s** / 41.88s => _this part has been heavily upgraded_
- Printing the purge tower : **2.50s** / 44.38s => _now it is no more needed_

Total time with 400mm³ purge 1mm³ tower : **44.38s**
Total time with 400mm³ purge  : **41.88s**

Note : after other changes, now I can remove the purge, it does not make the print look nicer anymore

HobinBambu, 2023-04-27:
Hi Dzzd
    Thank you for your PR and related discussion link."place holders" is also the way we are trying, but there are still some cases(include the "cut") that cannot be solved by "place holders",For example, when the filaments are all configured as "Generic PLA", the parameters they need are still very different.
1)Please update the firmware released yesterday https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history,
This version also supports configuring the purging speed and temperature when the filament run out (executed in the T command)
[image]
2)The ”cut“ in the T command uses the sensor, which cannot be realized by using ordinary G0/1.
3) during loading/unloading,before the filament pulled out of the tool head or when it fed into the tool head,there are many steps used to ensure the reliability(won't cause clog, new filament can be grabbed reliably,etc),this will take a few seconds.
Thanks.

DzzD, 2023-04-27:
Nice, any chance to have documentation on M620.1 and other custom g-code you handle ?

DzzD, 2023-04-27:
It would be nice to be able to contribute to the g-code outside of a json file in a plain text file on the repository, comparaisons and pull request as well as colaboration would be facilitated

adripo, 2023-07-18:
How can we help to solve this issue and related PR?

wkupike2000, 2023-09-22:
It seems like this got lost in the shuffle. Any updates on improving this speed?

EStreet706, 2023-10-16:
What is also needed is this.  adding in the code from this project.  https://www.printables.com/model/582382-bambulab-profile-for-up-to-60-purge-reduction  for the change filament it is simply a 20mm retract BEFORE cutting and a 20mm extrude AFTER cutting.  This saves a ton of waste.  Also for this posted project you need to change flushing volume and we have it set down to 0.40 which is 60% reduction in waste.

DzzD, 2023-10-20:
> What is also needed is this. adding in the code from this project. https://www.printables.com/model/582382-bambulab-profile-for-up-to-60-purge-reduction for the change filament it is simply a 20mm [...]

This is already something I have taken into account in another Issue : 
https://github.com/bambulab/BambuStudio/pull/1660#issuecomment-1750302177