bambulab/BambuStudio #1662: [Feature Request] Allow use of NFC stickers to save filament settings for a spool.

LayerCakeMakes, 2023-04-23:
NFC stickers are widely available for cheap, I would like to be able to stick these to any kind of spool and then have the printer safe changed settings (filament type, color, and temperature etc.) to them once I synchronize with the AMS after setting the information through the Bambu Studio UI.

sodachen22, 2023-04-25:
It is not supported to write back the settings to the NFC stickers? And also it is not easy to seated the stickers in non-bambu filament. There may need stand-alone labels stickers. But if there were, how much would you be willing to pay for a pair of stickers?

Majestic7979, 2023-04-25:
Absolutely! I want to be able to tag my spools and save the settings back to the spools. This is really important for stock keeping, because sometimes it's important to dry the spools after a certain date to keep the filament supple. I had so many issues this week with having to use colors that I rarely use, the filament kept breaking inside the AMS and I had to take it all apart, the filament was too brittle and I didn't realize this until I woke up to failed prints because the filament was breaking inside the PTFE tube. Having a tag could help keep track of dates and then get reminders on the screen that the filament is potentially humid and can cause issues. I could of course place a sticky label on each spool but that is wasteful. With a respool tool I can reuse the spools with the tags multiple times therefore saving money on labels.

The AMS already has a RFID chip reader and programmer, so this is doable but Bambu Lab is preferring to lock it down. That is not cool, Bambu Lab. It takes a lot of time to dial-in settings for any filament and knowing there is a way to save it to a tag but you do not allow just makes me feel like you do not care about users but only about making money with sales of your own spools which are not at all going to cover all user cases. You should not use NFC as a differentiator to sell your own filament. Instead focus on adding features to the printer to set you apart in the market. You are starting to act like printer manufacturers that rip customers off with printer ink. Except they give their printers almost for free and then charge more for the filament. Your printer is not cheap (neither should it be) so stop locking features down to your own filament.

LayerCakeMakes, 2023-04-25:
> It is not supported to write back the settings to the NFC stickers? And also it is not easy to seated the stickers in non-bambu filament. There may need stand-alone labels stickers. But if there were, how much would you be willing to pay for a pair of stickers?

I would expect to use standard off the shelf NFC/Mifare stickers those vary between 0.20~0.50 € getting even cheaper when ordered in bulk.
The placement should not be an issue we could just 3d print a jig that emulates the rollers of the AMS and has a hole where to stick the NFC sticker at the right distance. Something like this. 
[image]

Majestic7979, 2023-04-25:
> It is not supported to write back the settings to the NFC stickers? And also it is not easy to seated the stickers in non-bambu filament. There may need stand-alone labels stickers. But if there were, how much would you be willing to pay for a pair of stickers?

You are speaking like an executive from Hewlett Packard here, asking "how much are you willing to pay". Not everything has to make you money, you know?! The hardware is there. You can turn this into a massive selling point. Gain some good karma with the Open Source people by opening your AMS to standard tags. And if the hardware can read then it can also write. Is Bambu Lab planning to release inkjet printers soon and charge £50 per ml of ink? Because that's like, kinda what you're trying to do with your AMS. You're making the printer dumb unnecessarily. It's a 3D printer, man. Your company cannot cover all the filament options in the world. The hardware is there just make it open already. You should make money from the hardware not the supplies. You're either a hardware company or a printer supplies company. What is it? Ok it's cool that you offer branded filament but this has absolutely nothing to do with the NFC sticker. You are killing a selling point by doing this. As the guy above said, adapters can help positioning the sticker and there's nothing wrong with allowing simple MiFare tags.

This is a scan from an empty BL spool:
[image]

And this is a bunch of standard MiFare classic tags I could be using to make my life easier right now if Bambu Lab stopped trying to monetize every idea. The AMS costs over £300, and it's mostly just plastic. Stop trying to nickel and dime customers and open the system up already! You don't have to open source it but at least allow the user to write their spool information to their own spools and tags!

[image]

JelbertHoltrop, 2023-04-27:
I can understand that BambuLabs will not open up their internal security for the printer. However there is a work around. Bambu could create an open format next to their proprietary RFID tags. Then when detecting a spool also scan for this other format after the bambu tag read failed. And if it is a timing issue make a menu option to enable scanning for external tags and the user then must accept the extra lag. 

If the readers can only read it is to the user to add the appropriate data to the tag.

LayerCakeMakes, 2023-05-05:
I now have a roll of Bambu Lab PLA Basic in black were the NFC seems to be broken no matter what I try different slots etc. they seem to be broken. If I could write my own stickers I could fix this but now I'm stuck without the information showing for the AMS.

phranck, 2023-07-06:
In addition to this topic, there is something else that is very important to me: For an open communication between Bambu Lab employees and their many customers, also here on GitHub, I think it is first and foremost important that Bambu Lab employees regularly speak up here. Even if it's just a "Hi! Yes, we understand the problem, and we're working on it." **Feedback is SOOO important!**

Furthermore, I would like to see Bambu Lab employees here have a properly filled out profile with a picture as well as information that tells me who I am dealing with. The same goes for the Bambu Lab forum, of course. And ideally the profile information is identical.

My two cents.

mplacona, 2023-10-02:
Any updates from Bambu on this one?

wavelov3r, 2023-11-06:
+1 for Customized NFC capabilities! .. or it's a marketing move?

drc85, 2024-01-10:
would also be good for the enviroment lol

lets hope one at least day they will let us rewritte them with third party fillaments

Agoutii, 2024-02-03:
Given the markup the Bambu has on their filament rolls (since they cost nearly double what the manufactures sell the same rolls as), and the effort they put into preventing people using custom RFID tags (encryption and magic card detection) I think this is sadly very unlikely. Filament sales driven by AMS convenience is too big of a cash cow to risk consumers shopping outside the ecosystem.

Of course, for many such as me, these sorts of tactics really turn me off and make me want to avoid Bambu filament completely. Ethical conduct is something we should all support.

Majestic7979, 2024-02-03:
> Given the markup the Bambu has on their filament rolls (since they cost nearly double what the manufactures sell the same rolls as), and the effort they put into preventing people using custom RFID [...]

This is a silly argument, with due respect to you of course. The argument itself is ridiculous because let's take Nespresso coffee machines... You can go buy third party capsules but the vast majority of people buy original Nespresso pods. This is because of effective marketing and the average consumer just wanting to trust the manufacturer. If Bambu has a better product (filament is chemically more stable, doesn't need to be dried under reasonable humidity conditions and doesn't get brittle and break inside the AMS after a few months, just to name a few reasons) then people will buy the Bambu filament instead.

Or they could just lower the price and kill the competition because of the RFID. Because seriously, the roll of orange filament I got with my Carbon became brittle while being stored in the exact same condition I stored my Copymaster3D filament from Technology Outlet (in the UK). The Copymaster filament has never been dried, never been stored in special controlled conditions, and every time I print, it's perfect. I have several colors of it.

The Copymaster color range is much higher than the Bambu range and while they are also out of stock like the Bambu filament, the price is basically lower and the filament is better. I don't care about RFID if the filament itself is garbage, and in Bambu's case it seems to be. That orange filament from Bambu broke inside the AMS tubes, I had to take the whole AMS apart to fish out the broken pieces. There's no RFID that can save you from that frustration. So nothing will make me buy Bambu filament. But I did pay for the AMS in full, over £300 for it! I want to be able to use it fully with whatever filament I have on hand.

Besides, Bambu Lab does not make all filament materials and colors that I might need, so now I have a crippled device because they won't allow me to just use off the shelf tags with my third party spools. This is petty and ridiculous, it's taking a leaf out of Hewlett-Packard's book with them crippling printers when users insert third party cartridges. Bambu's strategy does seem to slowly become more apparently over time - they are probably making a loss on the printer side hoping to cannibalize sales of filament. I don't like that one bit.

I will give BL a year or so to make the AMS open and for us to be able to program filament data into NFC - the foundation has been laid now with custom filaments in the latest firmware. Now Bambu needs to stop being so anticonsumerist and open the RFID system up. I would love to press a button on Bambu Studio that would write the filament data to a virgin NFC tag I put on a roll of filament, and have it subsequently behave from there like a Bambu RFID spool, with tracking and all. There's no reason why Bambu Lab shouldn't do this.

Majestic7979, 2024-02-03:
> I can understand that BambuLabs will not open up their internal security for the printer. However there is a work around. Bambu could create an open format next to their proprietary RFID tags. Then [...]

This has NOTHING to do with security. Don't be fooled. The tags are basically a database with fixed fields and changing values. The vendor lock is nothing more than a business strategy. The firmware has protections in place to not go above a certain hotend temperature, not go above a certain heatbed temperature, etc. So even if a malicious actor tried to program values that are too high for a filament, say 500 degrees Celsius, the printer will never go above 300 degrees Celsius because that is the limit hardcoded in firmware. Bambu is locking this up because they don't want the competition selling spools with RFID. This is Bambu's strategy and it's clear because it's the differentiator. Filaments are pretty much all the same chemically, so PLA from Vendor A will be like Bambu PLA broadly.

Bambu's "security" is the same "security" that HP is using for their printers now, where the printer stops working if you put in a third party cartridge. HP recently said "cartridges can include malware that can be installed when it's inserter in the printer", however, this was under lab conditions and HP can always harden their firmware more so there's no code execution while reading cartridge chips. Bambu is capable of doing the same if security is a concern.

This is nothing more than a business strategy: Bambu wants to sell you a cheaper printer while making a ton of money on consumables that the user must continue to purchase because the whole point of 3D printing is needing raw material, which can be infinite. But that means Bambu Lab should basically give you the AMS for almost free.

This isn't the case, it is mostly a piece of injection moulded plastic with a few custom boards and motor for £300. That's an insane price for what it is and it should be opened up so that with the newer firmware on the X1 and Bambu Studio enabled now for custom firmware parameters the user should be able to program tags on their own spools. If Bambu wants to sell more filament they should lower the cost. Their "filament club" idea is ridiculous. They should just simply lower the price across the board for every customer, make sure they always have stock of all colors and materials, give great bulk discounts, and include free and fast shipping. I can get great quality PLA (better than the Bambu Basic PLA) from the UK with next-day shipping in a bigger range of colors from another company, if Bambu thinks I'll pay more and get less just because of RFID they're crazy.

secretman28, 2024-08-14:
lets bump up this, we should be ablee to do this ... rfid class1 gen2.... here my hint for you guys, more later on lol

littlejackal, 2024-09-06:
+1 for this feature

killsecurly, 2024-10-01:
Like others say, IT'S FOR MARKETING. There's only Bambu filament that auto RFIDs with the AMS. That's their selling point for the AMS. I buy Bambu filament for one reason: I don't want to manually edit the filament in Studio.

secretman28, 2024-10-08:
now think you can put any brand and it will work the same way...

littlejackal, 2024-10-11:
> Like others say, IT'S FOR MARKETING. There's only Bambu filament that auto RFIDs with the AMS. That's their selling point for the RFID enabled spools. I buy Bambu filament for one reason: I don't want to manually edit the filament in Studio.

Bambu is entirely within their rights to close this as a wontfix based on the above, if that is their position. Until they do, it's entirely reasonable to ask for this as a feature request. You'll never get what you don't ask for :)

secretman28, 2024-10-11:
> Bambu is entirely within their rights to close this as a wontfix based on the above, if that is their position. Until they do, it's entirely reasonable to ask for this as a feature request. You'll [...]

amen

3DJupp, 2025-01-26:
Similar Situation fot Anycubic, seems like the AMS/ACE (Pro) manufacturers are trying to increase sales with this NFC Tag equipped Material. I think its not really worth the price difference, but thats just my personal aspect. 

Anyways, for the ACE Pro there are similar attempts, the color codes are available. I hope that those two and probably more companies open up their systems to 3rd party material with NFC-Tage. But whats probably will happen: If anycubic will open the ACE, Bambu has to follow for the AMS/AMS 2 Pro.

[Link to a discussion on Reddit](https://www.reddit.com/r/anycubic/comments/1g047ad/ace_pro_nfc_tag_research_create_own_tags/)

Skye-Bambulab, 2026-11-16:
Hello, thank you for sharing your feedback. We understand the use case and will keep this as a feature request for future evaluation.

However, we would like to clarify that we currently have no plans to develop this feature. Given our closed-source system architecture and current development roadmap, this is not a priority for us in the short term. Should there be any changes to our plans in the future, we will update the community accordingly.

We appreciate your patience and your continued support.