bambulab/BambuStudio #1663: Please add a "Fast Start" option

DzzD, 2023-04-23:
Often, all printer preparation before print is useless, it would be great to have a button or a checkbox to enable a "Fast Start" option. 

Once clicked or checked, a warning should appear :
- You have requested a fast start, please be sure that :
- the nozzle is higher than the bed
- you printing plate is correcly set in place
- you did not forget any part on the printing bed

With "Fast start", the printer should only do at start before printing : 
- a homing
- a prime line

bambu123, 2023-04-25:
[image]

You can disable these two for saving time.

DzzD, 2023-04-25:
Thank, I know that, but there is still tons of preparation done by the printer, would not it be possible to just do a homing, a prime line and then start ? (using maybe a new checkbox as in your printscreen.

Majestic7979, 2023-04-25:
I initially nodded in agreement but now I think of it, this is not a wise idea. What takes most of the time is the bed levelling and that's something you really should not be wanting to skip. If there is a bed height discrepancy, maybe because you are using a different plate than the one in the slicer, then the printer won't know and you could have nozzle or print plate damage from scraping. Also, wasted filament if it delaminates. Unchecking the two boxes above does the bare minimum to the process. Honestly it's not worth it - I know what it feels like when someone says your idea isn't appropriate, it sucks, you want choice, but if the choice is to potentially cause damage I think they obviously want to limit the liabilities.

bambu123, 2023-04-25:
There remain nothing optional after turn off Bed leveling and Flow calibration.
- Heat
- Home

DzzD, 2023-04-25:
> I initially nodded in agreement but now I think of it, this is not a wise idea. What takes most of the time is the bed levelling and that's something you really should not be wanting to skip. If [...]

?! I rarely use bed levelling until I change my plate (the kind of plate), and that's true for all my printer, you really do a bed leveling each print ? you're madman :), 

Also I told about a Warning for this option, but frankly, most pre-ops are for noobs, I know when I should do a bed levelling, I know when I dont, I know if I leave something  one my plate or if my bed is too high, etc... having choice would be good and if a warning/disclaimer if you are fear ;)

DzzD, 2023-04-25:
May I try replacing the whole g-code by a G28 single command ? or is there a risk elsewhere ?

Majestic7979, 2023-04-25:
> > I initially nodded in agreement but now I think of it, this is not a wise idea. What takes most of the time is the bed levelling and that's something you really should not be wanting to skip. If there is a bed height discrepancy, maybe because you are using a different plate than the one in the slicer, then the printer won't know and you could have nozzle or print plate damage from scraping. Also, wasted filament if it delaminates. Unchecking the two boxes above does the bare minimum to the process. Honestly it's not worth it - I know what it feels like when someone says your idea isn't appropriate, it sucks, you want choice, but if the choice is to potentially cause damage I think they obviously want to limit the liabilities.
> 
> ?! I rarely use bed levelling until I change my plate (the kind of plate), and that's true for all my printer, you really do a bed leveling each print ? you're madman :),
> 
> Also I told about a Warning for this option, but frankly, most pre-ops are for noobs, I know when I should do a bed levelling, I know when I dont, I know if I leave something one my plate or if my bed is too high, etc... having choice would be good and add a warning/disclaimer if you are fear ;)

No, I don't do the bed levelling. The printer does. This is not an Ender where you have to manually adjust screws. You're being kind of dramatic... it's 5 extra minutes for always perfect first layer. It's not a big deal.

DzzD, 2023-04-25:
Man you are just seeing the whole world by your small windows, people use 3d printers in severals differents ways

Majestic7979, 2023-04-25:
> It is up to the Git project managers, so basically Bambulab to choose what they want to prioritize, as any opensourced project, not me, not you.
> 
> Prioritisation in software mays often seems not always logical, it is all a matter of ROI, small but not so usefull & easy modification can have a higer ROI than big & complexe modifications very usefull,
> 
> This is all depending on the criterias they choosen and the final ROI

I didn't need your mansplaining. I know it. Read again to understand my point - the more feature/bug requests submitted the more competition for their engineering time. Your request is for something that is nagging you at best, it's not a bug nor is there anything missing. I am not gonna come back to waste my time arguing about it. They will prioritize as they see fit, being that it's their business, but there are many more important things required right now than adding a checkbox, especially when they replied above that they can't shorten the time further.

DzzD, 2023-04-25:
Welcome to the fantastic world of Open source :)

Sometime I do myself, and I may try to do this one if i got enough free time, and despite @bambu123 answer, it can be accelerated...

Acelya-9028, 2024-08-25:
> > Man you are just seeing the whole world by your small windows, people use 3d printers in several different ways
> > **EDIT:** you should maybe study a bit more all those things, you seems anxious wich I dont understand why, and your question look very innocents
> 
> No they are absolutely normal questions. Bambu Lab has a set number of developers like any company. They have to split their time between priorities. As customers here we are all fighting for their time to get the features that we want. If they give you this feature, I have to wait for longer to get mine. So I want to know why your feature should be prioritized. Making you wait 5 minutes per print gives you the advantage of always having a perfect first layer, which is hardly something you can complain about. But when I am unable to update my printer firmware because my printer isn't connected to the cloud and they should build a way to upload firmware images via SD or LAN mode, then this is clearly detrimental to myself and other users who need this feature (I am not alone, you can see the other requests in here). So in terms of priority I don't think yours should be prioritized because it causes you no issues whereas in my case and other people's cases our printers are not getting bug fixes if we operate offline.

You're a clown omg.

xeliuqa, 2025-06-08:
Tell you what, i was forced to buy an A1 mini just to skip the annoying checks on the X1C, because i do a lot of small test print before final product.
I get it why they are there but i also agree with the fast start at our own risk. Printing with PLA-CF gives you a warning that we ignore, why not a fast start?
There's a lot of mods around but i don't feel comfortable using them.

So yes im in favour of Fast start button with a warning pop-up.