bambulab/BambuStudio #1665: Studio AppImage crashes when clicking "play" on the camera on Fedora 37

IOExceptional, 2023-04-24:
The app closes out with no error dialogue after I click "play" on the camera

Steps to reproduce the behavior:
1. Open Bambu Studio 1.5.0-hotfix1 OR 1.6.0
2. Click on "Device" tab
3. Select printer from the left menu dropdown
4. Click "Play" below the black box 
5. App crashes

**Expected behavior**
Instead of 5., it should display the output from the camera

**Desktop (please complete the following information):**
 - OS: Fedora 37
 - Bambu Studio 1.6.0-preview1 AND 1.5.0-hotfix1 (using the AppImages)

lanewei120, 2023-04-24:
thanks for reporting this issue

please help to check the libjpeg.so's version on your system
for example, on my laptop, it is libjpeg.so.62
I can find /lib64/libjpeg.so.62 on my laptop

lanewei120, 2023-04-26:
@IOExceptional 

could you help to check the libjpeg's version on your Fedora

IOExceptional, 2023-04-26:
@lanewei120 I have `libjpeg.so.62.3.0`

[image]

lanewei120, 2023-04-27:
so it should be the issue mentioned in https://github.com/bambulab/BambuStudio/issues/1189
currently the libjpeg version used in bambustudio is jpeg8
it has conflicts with jpeg6 on your system

we will upload a fedora version using jpeg6 in the future versions

lanewei120, 2023-04-27:
another solution is to upgrade libjpeg to jpeg8
however, as We don't have Fedora 37, or 38 at our side
we can not confirm how to do this

lanewei120, 2023-05-05:
we have downgraded the jpeg version to jpeg62
could you try the newest image on Fedora?https://github.com/bambulab/BambuStudio/releases/download/v01.06.02.04/Bambu_Studio_linux_fedora_v01.06.02.04-20230427094209.AppImage

Matthew-Beckett, 2023-05-16:
> we have downgraded the jpeg version to jpeg62 could you try the newest image on Fedora?https://github.com/bambulab/BambuStudio/releases/download/v01.06.02.04/Bambu_Studio_linux_fedora_v01.06.02.04-20230427094209.AppImage

I have tested this on Fedora 37 and it's now working properly for me.

WayneManion, 2023-08-01:
I am running Arch. I am running the AUR package (built from source) from here: https://aur.archlinux.org/packages/bambustudio-git

The application doesn't crash when I click the play button to view the camera stream, but I can't see any video either. It just says:

Loading Failed [2]!

The gdb logs from running the built-from-source version of the app is here: https://pastebin.com/bhUK8JAh

KOGJet, 2024-06-16:
> > we have downgraded the jpeg version to jpeg62 could you try the newest image on Fedora?https://github.com/bambulab/BambuStudio/releases/download/v01.06.02.04/Bambu_Studio_linux_fedora_v01.06.02.04-20230427094209.AppImage
> 
> I have tested this on Fedora 37 and it's now working properly for me.

Could you do this with the latest release on Ubuntu? Im new to this and hoping it could be my fix.

lanewei120, 2024-06-17:
@KOGJet for ubuntu, please try the ubuntu image instead of the fedora image
and do you have issue with ubuntu image?

KOGJet, 2024-06-17:
I tried both the fedora and Ubuntu and both either crash or show a error saying this device cannot handle more conversations.

lanewei120, 2024-06-17:
maybe your issue is another one
how about the flathub image? https://flathub.org/apps/com.bambulab.BambuStudio

KOGJet, 2024-06-17:
how do I install this into linux ? it didn't download as an app image but a ref pack

KOGJet, 2024-06-17:
I get this error when running the script to download through flatpak[ ](error: Flatpak system operation ConfigureRemote not allowed for user)

lanewei120, 2024-06-17:
[image]

KOGJet, 2024-06-17:
yes I tried to run those lines and got the [](error: Flatpak system operation ConfigureRemote not allowed for user)). How do i fix that?

lanewei120, 2024-06-17:
let's report an issue here
https://github.com/flathub/com.bambulab.BambuStudio/issues