bambulab/BambuStudio #1667: Appimage for v01.06.00.65 (ubuntu) does not work

abhijitmajumdar, 2023-04-24:
**Describe the bug**
When trying to run the Appimage for the latest v1.6 release on ubuntu 20.04, it does not start with the following error
```
/tmp/.mount_Bambu_YB6oI1/bin/bambu-studio: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by /tmp/.mount_Bambu_YB6oI1/bin/bambu-studio)
/tmp/.mount_Bambu_YB6oI1/bin/bambu-studio: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/.mount_Bambu_YB6oI1/bin/bambu-studio)
/tmp/.mount_Bambu_YB6oI1/bin/bambu-studio: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /tmp/.mount_Bambu_YB6oI1/bin/bambu-studio)
/tmp/.mount_Bambu_YB6oI1/bin/bambu-studio: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/.mount_Bambu_YB6oI1/bin/bambu-studio)
/tmp/.mount_Bambu_YB6oI1/bin/bambu-studio: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /tmp/.mount_Bambu_YB6oI1/bin/bambu-studio)
/tmp/.mount_Bambu_YB6oI1/bin/bambu-studio: /lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.13' not found (required by /tmp/.mount_Bambu_YB6oI1/bin/bambu-studio)
/tmp/.mount_Bambu_YB6oI1/bin/bambu-studio: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /tmp/.mount_Bambu_YB6oI1/bin/bambu-studio)
```

For reference I am trying to run this appimage `Bambu_Studio_linux_ubuntu_v01.06.00.65_20230417141721.AppImage` downloaded from the [release page](https://github.com/bambulab/BambuStudio/releases/tag/v01.06.00.65). I am still able to run the older version Appimage `BambuStudio_ubu64.AppImage` which was extracted from [this](https://github.com/bambulab/BambuStudio/releases/download/v01.05.00.61/Bambu_Studio_linux_ubuntu-v01.05.00.61-20230314200047.zip) obtained from https://github.com/bambulab/BambuStudio/releases/tag/v01.05.00.61

**3mf File for This Bug**
None

**To Reproduce**
Steps to reproduce the behavior:
1. On a system with Ubuntu 20.04, download the Appimage from https://github.com/bambulab/BambuStudio/releases/download/v01.06.00.65/Bambu_Studio_linux_ubuntu_v01.06.00.65_20230417141721.AppImage
2. make it executable `chmod +x Bambu_Studio_linux_ubuntu_v01.06.00.65_20230417141721.AppImage`
3. Run it `./Bambu_Studio_linux_ubuntu_v01.06.00.65_20230417141721.AppImage`
4. See error as shown above

**Expected behavior**
No errors, and Bambu Studio should start

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Version 20.04

lanewei120, 2023-04-25:
the latest ubuntu image is built on github using ubuntu-latest
the previous ubuntu image is built on github using ubuntu-20.04

it seems the glibc is not compatible

we will try to find a solution

lanewei120, 2023-04-25:
could you try this image for a try
https://github.com/bambulab/BambuStudio/suites/12456406244/artifacts/663491270

I just switched the building server back to ubuntu 20.04(previouly there is a building server error on github)

abhijitmajumdar, 2023-04-25:
> could you try this image for a try https://github.com/bambulab/BambuStudio/suites/12456406244/artifacts/663491270
> 
> I just switched the building server back to ubuntu 20.04(previouly there is a building server error on github)

Tried the image you referenced, unfortunately had the same error, see logs below

```
abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio/new$ ll
total 81M
drwxr-xr-x 2 abhijit abhijit 4.0K Apr 25 14:53 ./
drwxr-xr-x 5 abhijit abhijit 4.0K Apr 25 14:51 ../
-rw-r--r-- 1 abhijit abhijit  81M Apr 25 14:51 BambuStudio_Linux.zip
abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio/new$
abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio/new$ unzip BambuStudio_Linux.zip
Archive:  BambuStudio_Linux.zip
  inflating: BambuStudio_ubu64.AppImage
abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio/new$ chmod +x BambuStudio_ubu64.AppImage
abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio/new$ ./BambuStudio_ubu64.AppImage
/tmp/.mount_BambuSsSPZUB/bin/bambu-studio: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by /tmp/.mount_BambuSsSPZUB/bin/bambu-studio)
/tmp/.mount_BambuSsSPZUB/bin/bambu-studio: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/.mount_BambuSsSPZUB/bin/bambu-studio)
/tmp/.mount_BambuSsSPZUB/bin/bambu-studio: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /tmp/.mount_BambuSsSPZUB/bin/bambu-studio)
/tmp/.mount_BambuSsSPZUB/bin/bambu-studio: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/.mount_BambuSsSPZUB/bin/bambu-studio)
/tmp/.mount_BambuSsSPZUB/bin/bambu-studio: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /tmp/.mount_BambuSsSPZUB/bin/bambu-studio)
/tmp/.mount_BambuSsSPZUB/bin/bambu-studio: /lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.13' not found (required by /tmp/.mount_BambuSsSPZUB/bin/bambu-studio)
/tmp/.mount_BambuSsSPZUB/bin/bambu-studio: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /tmp/.mount_BambuSsSPZUB/bin/bambu-studio)
abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio/new$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.4 LTS
Release:        20.04
Codename:       focal
abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio/new$ uname -a
Linux 6NX1TQ-Ubuntu 5.14.0-1045-oem #51-Ubuntu SMP Mon Jul 4 06:41:22 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

lanewei120, 2023-04-26:
so could you check your glibc version?
the built scripts of (https://github.com/bambulab/BambuStudio/suites/12456406244/artifacts/663491270) is the same with before

if you can run v1.5 and can not run the above one
most likely it is caused by github's building server upgraded glibc libraries

abhijitmajumdar, 2023-05-10:
> so could you check your glibc version? the built scripts of (https://github.com/bambulab/BambuStudio/suites/12456406244/artifacts/663491270) is the same with before
> 
> if you can run v1.5 and can not run the above one most likely it is caused by github's building server upgraded glibc libraries

Sorry I could not get back earlier. Seems like my glib version is a minor version below what is required by BambuStudio. Here are some logs

```bash
abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio$ ldd --version
ldd (Ubuntu GLIBC 2.31-0ubuntu9.9) 2.31
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio$ strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBCXX_3.4.26
GLIBCXX_3.4.27
GLIBCXX_3.4.28
GLIBCXX_DEBUG_MESSAGE_LENGTH

abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio$ apt-cache policy g++
g++:
  Installed: 4:9.3.0-1ubuntu2
  Candidate: 4:9.3.0-1ubuntu2
  Version table:
 *** 4:9.3.0-1ubuntu2 500
        500 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status

abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio$ apt-cache policy gcc
gcc:
  Installed: 4:9.3.0-1ubuntu2
  Candidate: 4:9.3.0-1ubuntu2
  Version table:
 *** 4:9.3.0-1ubuntu2 500
        500 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
```

Is there any other way to run the Appimage without upgrading my glib?

lanewei120, 2023-05-10:
> Is there any other way to run the Appimage without upgrading my glib?

I am afraid currently I have no idea on this
it seems upgrading the glibc is an easy way, just as https://askubuntu.com/questions/1393285/how-to-install-glibcxx-3-4-29-on-ubuntu-20-04?

Obscuriosity, 2023-05-10:
Having the same issue, also on Ubuntu 20.04. likewise my ldd version is -
ldd (Ubuntu GLIBC 2.31-0ubuntu9.9) 2.31
The preceding link removed the following error.
/tmp/.mount_Bambu_S8lazF/bin/bambu-studio: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /tmp/.mount_Bambu_S8lazF/bin/bambu-studio)

Obscuriosity, 2023-05-10:
Bambu_Studio_linux_ubuntu_v01.05.00.60_20230309155250.Appimage
Seems to work for me, It has opened and sliced a stl. I am printing it now, I'll let you know how it goes.

abhijitmajumdar, 2023-05-10:
> I am afraid currently I have no idea on this it seems upgrading the glibc is an easy way, just as https://askubuntu.com/questions/1393285/how-to-install-glibcxx-3-4-29-on-ubuntu-20-04?

upgrading g++ to 11 only removed part of the error before. The AppImage still fails to load with the following
```bash
abhijit@6NX1TQ-Ubuntu:~/Downloads/BambuStudio$ ./Bambu_Studio_linux_ubuntu_v01.06.00.65_20230417141721.AppImage
/tmp/.mount_Bambu_SwkEn1/bin/bambu-studio: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by /tmp/.mount_Bambu_SwkEn1/bin/bambu-studio)
/tmp/.mount_Bambu_SwkEn1/bin/bambu-studio: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/.mount_Bambu_SwkEn1/bin/bambu-studio)
/tmp/.mount_Bambu_SwkEn1/bin/bambu-studio: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /tmp/.mount_Bambu_SwkEn1/bin/bambu-studio)
/tmp/.mount_Bambu_SwkEn1/bin/bambu-studio: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/.mount_Bambu_SwkEn1/bin/bambu-studio)
/tmp/.mount_Bambu_SwkEn1/bin/bambu-studio: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /tmp/.mount_Bambu_SwkEn1/bin/bambu-studio)
```
and my `ldd --version` is still 2.31 (similar to what @Obscuriosity observed as well)

Also upgrading glib seems not a safe thing to do (https://www.reddit.com/r/linux4noobs/comments/otsgxg/comment/h6xkab6/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)? So it looks like these appimages are built to only be compatible with ubuntu 22+. It would be nice if the build scripts are modified to account for the default glib versions on different ubuntu versions (maybe the LTS versions in the past 5 years) by generating separate AppImages.

P.S. I am able to use the Docker image for the latest branch, however running it within docker, the GUI is really slow when changing the camera views. Not sure whats causing this, but I'd prefer running an AppImage instead

icholy, 2023-05-24:
I ran into the same issue. This version works though https://github.com/bambulab/BambuStudio/releases/tag/v01.05.00.61

icholy, 2023-06-04:
@abhijitmajumdar do you mind sharing how you ran this in docker?

abhijitmajumdar, 2023-06-05:
> @abhijitmajumdar do you mind sharing how you ran this in docker?

Run this
```bash
#clone
git clone https://github.com/bambulab/BambuStudio.git
cd BambuStudio
#build
./DockerBuild.sh
#fix(optional)
sed 's|docker run|docker run -h $HOSTNAME -v /tmp/.X11-unix:/tmp/.X11-unix --rm|g' DockerRun.sh
#run
./DockerRun.sh
```

TLDR: Just building the image and running it as listed [here](https://github.com/bambulab/BambuStudio/wiki/Docker-Run-Guide). The only changes I made is to add `-h $HOSTNAME -v /tmp/.X11-unix:/tmp/.X11-unix` to the the list of `docker run` params (and optionally a `--rm` to avoid polluting)

microwavesafe, 2023-06-29:
Same error here, I also don't think it's a good idea to update glibc. Work around is to use older release, which is ashame.

Is it possible to check the glibc version on the build platform?

lanewei120, 2023-06-30:
https://www.dropbox.com/s/3sr326af7ypxsex/BambuStudio_ubu64.AppImage?dl=0

I just uploaded this image built from my local ubuntu 20.04
@microwavesafe could you have a try to confirm whether it works or not?

abhijitmajumdar, 2023-06-30:
@lanewei120 I tried the appimage you generated, but that did not work for me still. I realized your logs show your system has GLIBCXX_3.4.30 (my system is limited to GLIBCXX_3.4.28). However, inspired from your effort, I generated the Appimage inside a container with ubuntu:20.04 as the base and that worked!

Realizing this works, I found a [commit made not long ago](https://github.com/bambulab/BambuStudio/commit/793c50b1a6033b2c92a93d9177e0b6c28f430b49) that shows that just before the release of BambuStudio v1.6 the base image on which the appimage is generated was changed from 20.04 to latest. I believe this is the reason its referencing a higher version of glib. [I created a PR](https://github.com/bambulab/BambuStudio/pull/1968) that duplicates the job to additionally create an appimage using a 20.04 base as well. If you allow for the workflow to run, I (and potentially everyone having issues) can verify the appimage generated by that works

lanewei120, 2023-06-30:
> @lanewei120 I tried the appimage you generated, but that did not work for me still. I realized your logs show your system has GLIBCXX_3.4.30 (my system is limited to GLIBCXX_3.4.28). However, [...]

Hi, that is just because the codes on master is not the newest
I have changed the building using 20.04 before
the commit is already in v1.6 branch
https://github.com/bambulab/BambuStudio/commit/b953fd0b4dc66328df84d295a011a19a72f0aa28

it will be merged to master soon

however this can not solve the issue as the ubuntu 20.04 on github had already updated too on March or April

abhijitmajumdar, 2023-06-30:
> however this can not solve the issue as the ubuntu 20.04 on github had already updated too on March or April

Ah! i see the issue now. Would it work if we switch to defining a docker workflow on github actions and use the [Containerfile](https://github.com/bambulab/BambuStudio/blob/693af0c98fdbce2b22cc0f3639db2e0c6d7803a9/Containerfile#L16) to build the Appimage?

lanewei120, 2023-06-30:
we can have a try,
from your test, is the performance the same after using docker building?

As I am not familiar with docker
could you give the details steps of the docker workflow?

microwavesafe, 2023-06-30:
@lanewei120 as mentioned the dropbox link build doesn't work. It seems strange that the 20.04 runner has a more updated glibc than the installed OS, it guarantees it won't work.

I'm not familiar with github actions, can you specify an even older OS say ubuntu 18.04? Otherwise the only solution is a custom docker container file.

lanewei120, 2023-06-30:
do you have an ubuntu 20.04 image only including the old glibc version?

maybe I can downgrade my ubuntu 20.04 to your version
and I will forbid the auto upgrading of ubuntu then

@microwavesafe @abhijitmajumdar

microwavesafe, 2023-06-30:
According to this page

https://launchpad.net/ubuntu/+source/glibc

20.04 will always install with glibc 2.31, so I would have thought a fresh installation of 20.04 will always have right glibc installed (2.31 / GLIBCXX_3.4.28)

abhijitmajumdar, 2023-06-30:
> As I am not familiar with docker
> could you give the details steps of the docker workflow?

Sure. I opted to use `podman` instead of `docker` to avoid dealing with root permissions. However, the workflow is similar.

- I use the base image as ubuntu-22.04 because that is what I had my self-hosted runner setup with, but this can be anything >= 20.10 (since podman is easily installable using apt for 20.10+)
- Then simply built the appimage inside an ubuntu 20.04 container using this [containerfile](https://github.com/bambulab/BambuStudio/blob/693af0c98fdbce2b22cc0f3639db2e0c6d7803a9/Containerfile#L16) already in the repo
  - The Containerfile already specifies a base image of ubuntu 20.04, so I did not need to change this
  - The Containerfile copies the repo for the current directory, which was cloned as per the currrent branch, and thus builds the latest
  - The major work is done in building this container with instructions in the Containerfile
  - The instructions to build the container are mentioned within the Containerfile [here](https://github.com/bambulab/BambuStudio/blob/693af0c98fdbce2b22cc0f3639db2e0c6d7803a9/Containerfile#L4C8-L4C8)
- After that I run the container to copy the appimage from inside the container to the host runner system
- Then upload the appimage

I did all this in my branch (see https://github.com/bambulab/BambuStudio/pull/1968/files), and that generated an Appimage that I could run on my system successfully. For everyone looking or a newer BambuStudio Appimage (based off of `master` branch with latest commit on June 14 2023), you can try https://github.com/abhijitmajumdar/BambuStudio/suites/13986070335/artifacts/780252615 (unzip, make the unzipped .Appimage file executable, and run).

NOTE: I had to make some changes to allow this to build on my self hosted server (see https://github.com/abhijitmajumdar/BambuStudio/actions/runs/5425761533). As a result, the PR will need to revert some changes before merging into this repo. @lanewei120 if you find this will be useful, please let me know I'll make those changes for your review in https://github.com/bambulab/BambuStudio/pull/1968

lanewei120, 2023-07-01:
> Sure. I opted to use `podman` instead of `docker` to avoid dealing with root permissions. However, the workflow is similar.
>
> * I use the base image as ubuntu-22.04 because that is what I had my [...]

thanks very much
it is useful, and we can go on with this PR

we will also have a try here

abhijitmajumdar, 2023-07-07:
> thanks very much
> it is useful, and we can go on with this PR

PR ready, waiting on approval for CI build so I can make I didn't miss a flag
ref: https://github.com/bambulab/BambuStudio/pull/1968

mihi4, 2023-07-20:
Hi! Thx for all the work making BambuStudio working vor 20.04 :-)
I'm on Mint 20.3 at the moment and just ordered a P1S.
Since 1.7Beta2 added support for the P1S - do you plan on building this version, too?
(Or would any P1P setting fit the P1S anyways?)

microwavesafe, 2023-07-20:
I ended up using OrcaSlicer (https://github.com/SoftFever/OrcaSlicer) (a fork of Bambulab Studio) which works perfectly on 20.04.

lanewei120, 2023-07-20:
we have also fixed the compiling issue under ubuntu 20.04 on github server

there is a image just built using 20.04
https://github.com/bambulab/BambuStudio/suites/14439598054/artifacts/815194371

please also have a try whether it can run or not on your ubuntu 20.04
@microwavesafe @mihi4

mihi4, 2023-07-20:
> we have also fixed the compiling issue under ubuntu 20.04 on github server
> 
> there is a image just built using 20.04 https://github.com/bambulab/BambuStudio/suites/14439598054/artifacts/815194371 (previously there is some issue on github, althrough we set to ubuntu 20.04, it still uses 22.04, which leads to the glibc version mismatch issue)
> 
> please also have a try whether it can run or not on your ubuntu 20.04 @microwavesafe @mihi4

Starts perfectly, thank you very much!
haven't sliced anything (printer is still in shipment *gg*), but so far everything looks great, I can select the P1S as my printer.

lanewei120, 2023-08-05:
already solved in v1.7.2.51, please have a try

closed 2023-08-05 by lanewei120