bambulab/BambuStudio #1670: Ability to rename a print in progress

aequasi, 2023-04-24:
I frequently forget to name a print/plate when I go to print it. It would be nice if we could update the print name while the print is in progress

walterwongbbl, 2023-04-25:
Hi.
What is the purpose of rename during the printing process. We need to evaluate this feature

aequasi, 2023-04-25:
"Housekeeping", for one. Just looking in bambuslicer when I forget to name something, it says that its printing "Untitled". Not super helpful. 

I'm also working on integrating with MQTT and Discord (also relevant to the Home Assistant codebases), and being able to have a status output showing the same thing ("Untitled") is also not super helpful

sodachen22, 2023-04-27:
> "Housekeeping" mainly. Just looking in BambuSlicer when I forget to name something, it says that its printing "Untitled". Not super helpful.
> 
> I'm also working on integrating with MQTT and Discord (also relevant to the Home Assistant codebases), and having a status output showing the same thing ("Untitled") is also not super helpful

For Home Assistant case, I think you already subscribe MQTT for the printer status, you can use subtask_name to identify printing job when the printing jobs have the same name.

aequasi, 2023-04-27:
Yeah, we do use `subtask_name` but if you don't change the print name before sending it to the printer, its `Untitled`, which doesn't help identify the print

aequasi, 2023-04-27:
That's not helpful, though. It's just a number.