bambulab/BambuStudio #1671: Add "exclude object" as per Klipper and "exclude region" as per Octoprint, to save multi object prints where one failed but the rest is fine.

f089435uj9gh43, 2023-04-24:
**Describe the bug**
This needs to be added to both the slicer and the printer.
Read this for info: https://www.klipper3d.org/Exclude_Object.html
Watch this to see it working: https://www.youtube.com/watch?v=QTwRZ_M159Q
This is also available in Octoprint but it's region-based rather than Object-based. I am asking for this in Object based form AND region-based form too if possible, with an option to choose which one I want for a particular print in the printer screen AND in bambu studio as well in the device tab.

Why:

Because 3D printing sucks... it's not perfect. If one object detaches from the build plate then you're gonna get spaghetti for the rest of the print even if the other objects are ok. This means you will waste filament, and all that spaghetti can also get in the way, get tangled and knock the previously-okay objects off the build plate.

This is a manual solution to let the user delete that object for the rest of the print but continue everything else, this minimizes the waste.

bambu123, 2023-04-25:
We doing something to support this.

Majestic7979, 2023-04-25:
> We doing something to support this.

Absolutely AMAZING! Is it coming on the next firmware upgrade? Can I ask if there is a release note or kanban board/roadmap for the firmware releases? It's a lot of anxiety waiting months for features, so knowing what is coming in advance and being aware it may or may not be in the official firmware would help users a lot.

bambu123, 2023-04-25:
If there are no surprises, the June version will have changes.

Majestic7979, 2023-04-25:
> If there are no surprises, the June version will have changes.

Is there a list of the changes? And when you say June version, does this mean the next firmware release is happening no earlier than June? I understand "no surprises" and that's fine.

bambu123, 2023-04-25:
We have released 1.6 currently.
The next version will be about two months later.

Majestic7979, 2023-04-25:
> We have released 1.6 currently. The next version will be about two months later.

Wait. I am talking about firmware here, not the slicer. You have not released 1.6 slicer yet, it's in beta. Are you saying that the slicer 1.6 beta can tag objects and is the current firmware on the X1c OTA version 01.04.01.00 (20230227) already able to delete objects from the screen? How? Where is this info on the wiki?

This feature requires a mix of firmware change in the printer as well as slicer change to tag the objects when doing the slicing. As far as I can see neither the firmware OTA version 01.04.01.00 (20230227) nor the slicer 1.6 beta have these implemented.

I want to ask for the third time do you have a list of features that you are planning to release in the next firmware? June is about 1 month away so surely the features have been locked already. You should be able to provide a list of upcoming features for the June firmware released "pending no surprises"

bambu123, 2023-04-25:
Yes, it requires the new OTA version of printer which is also under developing.
The feature list is internal only available now, sorry I can't provider it.

Majestic7979, 2023-04-25:
Ok no problem, at least it's good to know it's in the pipeline. Unfortunately seems like I won't be getting this ever, or until you decide to stop forcing OTA on users. I will not connect my printer online for any reason. So if BL doesn't make it possible to upload firmware on the LAN or via SD then this feature is impossible to install for my printer. I wonder why BL can't just initiate firmware updates via LAN mode!? I don't care about downloading the file. I don't mind if the slicer comes with firmware pre-loaded and if it sees that the printer connected via LAN mode is out of date, it could offer to update with the pre-loaded firmware. But I won't ever connect the printer itself to the internet to get updates. It's really sad that the printer will be forever stuck on the firmware that it came with.