bambulab/BambuStudio #1672: Ability to upgrade printer firmware using LAN mode or SD card

Majestic7979, 2023-04-24:
**Describe the bug**
I consider the update function partially broken if it's not possible to do locally - why? There's an SD slot. There's local mode/ftp running in the printer firmware already. If these things are there then it's possible, if it's not working then it's broken. If it's not implemented then it's broken. No reason not to do local firmware updates, this does not use any proprietary ports like cars or trucks. I am not here seeking argument with others. In my own eyes it's broken, others may share the same view or not, that's not up for debate, the printer should offer this option as it does not remove the cloud OTA but instead expands the ways in which an update can occur.

At the very, very least it should be possible to "send" an upgrade file to the printer via the slicer using LAN mode. But barring this for technical reasons, it should absolutely be possible to apply firmware upgrades entirely locally. Not gonna discuss the various reasons for this but privacy is definitely one. 

Not all businesses are willing or able to connect their printers online and I'd argue many businesses who have bought this printer didn't care about cloud, they cared about the physical features like CoreXY, enclosure, AMS, responsive screen etc. I want to have better performance and bug fixes and I don't want cloud features, not even to update the printer. Bambu Lab you have no excuse to stop this from happening and the printer is now behind on feature updates. I want the same firmware fixes and improvements that others are getting, but installed locally.

LayerCakeMakes, 2023-04-25:
I would also like this functionality in LAN mode. I honestly don't need any cloud features and this would be very helpful to keep my data local.

sodachen22, 2023-04-25:
> I would also like this functionality in LAN mode. I honestly don't need any cloud features and this would be very helpful to keep my data local.
Can I ask the reason that you want to keep all data local?

sodachen22, 2023-04-25:
Thanks for the feedback. OTA upgrading from cloud is not only a technical decision, but also a strategy decision.
In the original design, we want the cloud to give users a seamless experience, which includes printing, printing history and re-printing, OTA,
I'd like to have a survey that why you do not want to use cloud service. We can check if there is anything about the cloud that we can improve.

Majestic7979, 2023-04-25:
> Thanks for the feedback. OTA upgrading from cloud is not only a technical decision, but also a strategy decision. In the original design, we want the cloud to give users a seamless experience, which [...]

The strategy decision being what, exactly? As a business your mission is to make money AND to generate value for the user. You are NOT generating value by locking people up to the cloud. That should be optional. It's like forcing a user to connect to the internet to play Single-Player games. Remember what happened to SimCity? A fiasco. Because they insisted that cloud was required, but it was proved that it wasn't required, so Electronic Arts looked dumb. I am not alone to say I don't want your cloud no matter what you do. I am buying the hardware, and I like that Bambu Lab is making LAN mode (and hopefully improving on it).

But I know where you guys are going with your strategy, it's quite clear to me and I'll force you to confess now: You are offering cloud free-of-charge initially. Once you have thousands of customers all hooked into your system, you will in a few months talk about how the cloud operation is costing the business a lot of money and how Bambu Lab is gonna start charging a monthly fee to keep the service sustainable and operational. You will make people feel guilty because "the company will close if we don't do this", and then people will have no choice but to keep paying every month. THAT IS DISHONEST. I **KNOW** that this is your "strategy", which is why I am pushing for local functionality.

I would've been happy to pay more for the printer to get full local access for it, if that meant you made the amount of profit that you wanted to achieve. But to create a strategy where you shoehorn your existing customers into a subscription model down the line is NOT HONEST.

You are going to end up like Juicero and their stupid locked-down juicer. Look what happened to that company! I think most people who are not developers or Bambu Lab employees would agree that paying a monthly subscription to use a printer is stupid. Ok, if they want that then they can, it's their choice. But you are locking down the printer for firmware updates due to "strategy" and that's disingenuous. Are you selling the hardware to the user or renting it? Because if you want to exert full control over the hardware to the point that a local firmware update bothers you so much then that to me indicates you are not giving the user the printer when the money is exchanged, you are simply "renting" it. So if they don't want to use cloud features the product is crippled.

I won't measure any words here, that is SHIT, Bambu Lab. Change your strategy. From my perspective and I think I'm not alone, I don't want cloud anything on a 3D printer. I don't want it now, or in the future, no matter what you do. A WiFi printer that can do stuff with the slicer on the LAN is perfect for me. Locking down firmware updates is ridiculous.

LayerCakeMakes, 2023-04-25:
> > Can I ask the reason that you want to keep all data local?

Plain and simple privacy, why is it anyone's business what I print or how much I print.
Especially if I were to print contract work for other businesses there might be policies that would be violated by uploading to any cloud.
Printing and reprinting/history/ota all that could be done local there is no need at all for the cloud other than data harvesting and maybe remote access which I honestly don't need.

bl4ckm0onk3y, 2023-05-08:
> Thanks for the feedback. OTA upgrading from cloud is not only a technical decision, but also a strategy decision. In the original design, we want the cloud to give users a seamless experience, which [...]

Bought P1P, arrived today, cannot update to latest firmware thru cloud, always error, cannot update thru cloud, For that alone problem there would be a solution - update thru LAN or SD-Card = seamless experience, your cloud solution doesn't work.

Mystery86HH, 2024-02-02:
I bought two x1 carbons and two quidi x3 max for my company. I am working in our process engineering part in the company. Firmware updating the quidis with usb works great. The Carbone's I cannot update because of the company firewall... (Connecting to router failed at network test).
With hotspot on my private smartphone it will not connect to the Wi-Fi...

That's really bad. Please give a option to update the firmware with a micro sd-card.

madmodder123, 2024-12-29:
You can update via microsd card now if you have the following firmware versions or newer:

A1/A1 Mini: 01.04.00.00
P1 Series: 01.07.00.00 
X1/X1C: 01.08.02.00
X1E: 01.01.02.00

Majestic7979, 2025-01-07:
Closing issue, thanks Bambu Lab for listening to your customer who put the bread on your table. Now, please unlock the RFID. You are being offensive by locking it, you're acting like Hewlett-Packard and printer cartridges. RFID on spools does not need to be exclusive to your filaments. If you want to monetize just sell blank RFID tags so we can program them from the slicer for our own custome filaments.

closed 2025-01-07 by Majestic7979