bambulab/BambuStudio #1673: Suggestions

Scooter1010, 2023-04-24:
A couple recommendations and sorry if I just missed it and they do exist  
1. I would love a clone plate option in the slicer, I can clone the part and move it but cloning the plate with the orientation of the parts on it would be a time saver.
2. I would love to ignore color when using the ams rollover feature for filament (yes I know I could just label them all the same color) but would love if it was a feature to just ignore the color and only match type. (I had a print that it ran out using pla black and instead of switching over to another color it just sat there till I loaded a new reel in)

DzzD, 2023-04-25:
you shoul change your title to something more explicit, +800 issues, nobody will open all to see what is inside

bambu123, 2023-04-25:
Be good suggestions.