bambulab/BambuStudio #1675: Custom filament presets not available when switching to a different size nozzle

RebeccaStevens, 2023-04-25:
**Describe the bug**
When I switch from the 0.4 nozzle to the 0.2 nozzle, all the filament presets that I had previously set up are no longer available.

If I try to create the same profile again, using all the same setting, the slicer is perfectly happy with this. But if I try to save these settings with the same, it warns me that there is already a preset with that name that is incompatible with the current printer.
Why does changing the nozzle size make these presets incompatible?
Is it due to changing from the hardened steel nozzle to a stainless steel one? But all the filament presets in question are to do with non-abrasive filament so this shouldn't be an issue.

**3mf File for This Bug**
N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Create a custom filament preset
2. Switch to a different nozzle
3. Presets is no longer available

**Expected behavior**
The filament presets I've set up should be available for all the different nozzle options in the slicer.

**Screenshots**
[image]

**Desktop:**
 - OS: Windows
 - Version 10

bambu123, 2023-04-25:
Yes, currently. Custom filament preset is compatible with one nozzle only.
I think the issue is a feature request.

RebeccaStevens, 2023-05-02:
### Option 1

One way this feature could be added is to have an "overrides" section in the config where overrides for different "compatible printers" could be set. Actually, this overrides section could be implicit and calculated at runtime based on profiles that inherit from the base profile.

### Option 2

Another way this could possibly work is that when a user creates a new profile (`user-profile @base`), they don't just create a single profile that inherits from the base profile (`system-profile @base`), but they also create a new profile (`user-profile @0.2 nozzle`) of each of the profiles that inherit from that base profile (`system-profile @0.2 nozzle`).

One small issue with this is that the link between `user-profile @0.2 nozzle` and `system-profile @0.2 nozzle` would be lost. To avoid this, a three-way merge could be used where `user-profile @0.2 nozzle` would have to inherit from both `user-profile @base` and `system-profile @0.2 nozzle`.

I'm not sure if there would be an easy way to make this work for user profiles that already exist.

### Workaround

For now, you can override the `compatible_printers` in the filament's json config file.

```json
"compatible_printers": [
    "Bambu Lab X1 0.2 nozzle",
    "Bambu Lab X1 0.4 nozzle",
    "Bambu Lab X1 0.6 nozzle",
    "Bambu Lab X1 0.8 nozzle",
    "Bambu Lab X1 Carbon 0.2 nozzle",
    "Bambu Lab X1 Carbon 0.4 nozzle",
    "Bambu Lab X1 Carbon 0.6 nozzle",
    "Bambu Lab X1 Carbon 0.8 nozzle"
],
```

The only difference I've found between the `@base` and `@0.2 nozzle` profiles, is the `filament_max_volumetric_speed` settings (and maybe the `filament_start_gcode` - I haven't really looked at this).

adripo, 2023-07-19:
I see that filament configurations have `"inherits": "Generic PLA @BBL P1P 0.2 nozzle"` to associate a filament to a nozzle/printer, so it should be sufficient to implement multiple associations for the same filament.

As a workaround for now you can copy/paste the filaments (%APPDATA%\BambuStudio\user\YOUR_ID\filament) and change the value of `"inherits"` for each nozzle.