bambulab/BambuStudio #1677: "Independent support layer height" - this setting is garbage! It ruins prints.

Majestic7979, 2023-04-25:
I was losing my mind until I found out why my prints kept failing with a line that would make them snap right in the position of that line. This setting: "Independent support layer height" in the Support section of the slicer.

I was printing a bed with multiple parts and every single part had a defect on the exact same layer, this defect is basically under-extrusion meaning the bottom and top between this layer are separated, and as soon as you touch they separate cleanly, breaking the part in half! I wasted so much filament trying to find out why the printer was doing this, and it seems like disabling this setting fixed the issue. I was able to print without any breakages or layer under-extrusion.

tangketan, 2023-04-25:
I'm sorry for your loss, but I don't understand how "Independent support layer height" is related to your issue. Would you please share the 3mf? I'll turn off this param by default if this is a common issue.

Majestic7979, 2023-04-25:
Yes. Print this with and without the setting. Make sure that you paint supports and put support blockers where needed.

[Center_Discs_and_Spool_Axis_and_Nuts.zip](https://github.com/bambulab/BambuStudio/files/11321040/Center_Discs_and_Spool_Axis_and_Nuts.zip)

Majestic7979, 2023-04-25:
It seems that the printer doesn't manage support layer heights vs object layer heights very well, so the support gets a higher layer height value, when the printer moves back to the object it does not lower the nozzle, meaning it prints "in the air" over the object creating a layer gap, this gap then allows the object to break in half with the slightest touch, because it's not properly attached.

Look at the photo, two different objects on the same place, they both failed exactly on the same layer height. Looking at the slicer I could see the layer height difference, and that's what caused it. Once that setting above was turned off, objects printed beautifully, without issues. Only setting that changed was disabling that box. I tested this three times for a total of 6 tests, every single time that setting caused issue and disabling that setting made the pri
nt successful.
[image]

tangketan, 2023-04-27:
I'm sorry but can you send me a complete 3mf with painted supported? So I can exactly reproduce the issue.
I've tried twisting the support settings and there seems no supports generated at the layer where you got the adhesion issue.
[image]

Majestic7979, 2023-05-17:
Did you try to print the 3mf? I used the circle painting tool to paint green under the rim part of the left-most larger pieces on your image, where you have traditional supports on the far-left one. I made sure to leave some blocking space touching the cylinder across the middle of the parts, and I also blocked everything on the "cog" part at the bottom on the part where you can see the grid infill. The other two smaller cylinders I painted blockers internally because that's a female threaded hole so supports would cause issues.
When I used the setting "Independent support layer height" it caused delamination during the print. When I unticked that setting it printed fine exactly the same supports as before. Only the setting caused the issue.

tangketan, 2023-06-16:
Here are the printed samples, and also the 3mf.
[image]
[image]
[Center_Discs_and_Spool_Axis_and_Nuts.zip](https://github.com/bambulab/BambuStudio/files/11764863/Center_Discs_and_Spool_Axis_and_Nuts.zip)

Majestic7979, 2023-06-16:
The only difference is that I painted, are you using a different slicer version? I don't remember what version I was using but it was the latest at the time of posting. Maybe this metadata is included in the 3mf, I don't know... Let's close this issue for now and if it reoccurs in the future I will reopen it with comments. But mine was painted not auto supports. Thank you and nice flooring on the Bambu offices :-P

Majestic7979, 2023-06-16:
Oh and the reason why I painted the supports is actually showing in your sample:
[image]
Notice how there are leftover support even when you removed... I painted the rim above the gear to block support on that area, so that it would not touch the gear teeth

Majestic7979, 2023-06-16:
Check out this issue raised by another user, looks like this may be similar? https://github.com/bambulab/BambuStudio/issues/1927

tangketan, 2023-08-02:
That's possible. Now since #1927 is fixed, I believe this issue can be closed. Also pleased be noted that "Independent support layer height" option is now shown more obviously, so next time you can disable it more easily.
[image]

closed 2023-08-02 by tangketan

DerrillWMS, 2024-04-12:
"I was losing my mind until I found out why my prints kept failing with a line that would make them snap right in the position of that line ... **as soon as you touch they separate cleanly, breaking the part in half** ..."

I'm not being funny, but if whatever was actually causing this result can be fully identified, it is **a silver bullet**.  Heck, the "Support>Top Z distance" and "Support>Bottom Z distance" settings don't perform that well, and, frankly, not even breakaway support PLA (still can require a bit of post clean-up, especially because the support material is beige-white). I've performed hundreds of hours of tests on both, and even tried Matterhackers PVA water soluble support, but after numerous tests to try to get it to consistently stick to standard PLA, it's a no-go.

I just purchased Bambu's new water soluble PVA, and am anxious for it to arrive to test it. I'm hoping the fact that it has an RFID tag, and presuming Bambu has done exhaustive testing on settings for success, that the auto settings will work.

arturbacilla, 2025-04-17:
I know this thread is already closed but I belive it's the only I found that relates to problems I'm having. I'm also getting underextrusions on support interface layers but not that serious (my part isn't breaking apart). But my question is, how do this setting relates to Top/Bottom Z distance? They are co-dependant? In other words, if independent support layer heights is off, I need to set Z-dist always multiple of layer height or not necessarily?