bambulab/BambuStudio #1678: Unable to disconnect and "forget" WiFi network.

Majestic7979, 2023-04-25:
**Describe the bug**
It's currently impossible to "forget" the WiFi network that the printer has joined. I want to be able to disconnect and reconnect by using different credentials, manually, for example if I change the SSID password of a currently set network. I tried to tap on the network that the printer is connected to and it does not show any menus to forget or to disconnect. I can turn off the WLAN using the toggle but that doesn't disassociate the SSID on the printer. If I want to do this, I have to factory reset. There is also no way to manually enter the SSID name for a hidden network.

I do not want to use cloud or Bambu Handy, if this is something that can be managed on the app. I think all printer management functions must be available offline on the printer screen directly, including adding hidden SSID networks.

sodachen22, 2023-04-25:
1. Do you mean this forget option when the SSID is not current using?
[image]
2. One of the problem of manually entering SSID is multilange input when SSID is not in English. It is not easy for the printer to support multilingual input method on the touch screen.

Majestic7979, 2023-04-25:
> 1. Do you mean this forget option when the SSID is not current using?
>    [image]
> 2. One of the problem of manually entering SSID is multilange input when SSID is not in English. It is not easy for the printer to support multilingual input method on the touch screen.

No I mean the one that is being used! I want to be able to forget the current WiFi. And I can tell you 100% this is a case that might be needed because it happened to me today. For some reason the printer stopped communicating on the network. Even with a complete shutdown by removing the power cable and wait 5 minutes, the printer connects to the WiFi but it does not communicate. I managed to somehow forget the network and when I rejoined from scratch the printer was communicating again. But it was not possible to do this as you normally expect, to tap the current network and have the option to forget it. As you have just shown it's only possible to forget network that is disconnected. The network that is connected cannot be forgotten. If you turn off the WiFi then the networks all disappear from view.

You will be catering for a huge number of people if you just allow entering SSID with English characters. Doesn't the keyboard change based on language anyway? If someone wants to have a weird WiFi name with symbols that's their problem. But most people have just alphabet and numbers. You don't have to cater to a minority all the time.

sodachen22, 2023-04-26:
Yes, understood, forgeting current WiFi will be supported.
And we will also have more evaluation on hidden SSID input.

Majestic7979, 2023-04-26:
That's great. I want to expand on my feedback - I have a Ubiquiti UniFi network, three access points. For some reason the printer was preferring the access point with less signal quality! It was refusing to connect a lot. On the UniFi management page I can choose to lock my printer to a particular AP, once I locked to the AP closest to the printer, problem was resolved. So either the printer seems to be connected to the first AP it sees instead of picking the one with the best signal. As you know UniFi works similar to a mesh system so all APs have the same SSID, and it's up to the client to pick which one is the best to use.

sodachen22, 2023-04-27:
> That's great. I want to expand on my feedback - I have a Ubiquiti UniFi network, three access points. For some reason the printer was preferring the access point with less signal quality! It was [...]

When the printer power on or WiFi module is ON, it will connect to the AP which has the best signal quality. The signal quality is detected by the WiFi module. In this case, it looks like the criteria for signal quality are not consistent or the signal quality is not stable. Can you please have more testing to get more clue by turning on/off WiFi module?