bambulab/BambuStudio #1679: BUG: 1.6.0.65 Beta shows wrong external filament on Device Tab

moebis, 2023-04-25:
Shows wrong external filament on Device Tab, I currently have TPU set, and it shows as TPU on the printer, and most times it shows as TPU on the Device tab under the external filament, but about half the time it just decides to change it to PLA without me changing anything:

Screenshot:
[https://imgur.com/a/mBE4ht7](https://imgur.com/a/mBE4ht7)

walterwongbbl, 2023-04-25:
Hi.
Please provide the log file for the studio. I will try to find the reason.

moebis, 2023-04-25:
@walterwongbbl I emailed the log files to support.eu@service.bambulab.com

walterwongbbl, 2023-05-05:
> @walterwongbbl I emailed the log files to [support.eu@service.bambulab.com](mailto:support.eu@service.bambulab.com)

Thank you very much. I got the log file.

walterwongbbl, 2023-05-05:
Hi.
Does it mean that TPU is displayed on the printer, but PLA is displayed on Studio?
If the display on the printer and Studioare consistent, then the issue may be with the device. If they are not consistent, then the issue may be with Studio.

moebis, 2023-05-05:
> 

It was TPU on the printer and PLA on Studio. Since then (maybe 1.6.2 beta 2 of Studio) I haven't noticed it again.

walterwongbbl, 2023-05-05:
Hi.
According to your description, this should be a bug. 
But we did not find this issue during the testing process. 
I will keep an eye on this issue, and if I find and fixed it, I will notify you in this conversation.

walterwongbbl, 2023-05-31:
Hi.
We have reproduced this issue and will fix it in the next version

sodachen22, 2023-05-31:
> TPU

walterwongbbl, 2023-10-13:
Please download and use the latest version of Studio

closed 2023-10-13 by walterwongbbl