bambulab/BambuStudio #1682: All of the links from a setting to the documentation are broken for settings not new from Bambu Lab

julie777, 2023-04-25:
**Describe the bug**

One of the best features of Prusa Slicer is that every setting has a tool-tip with a brief description, and the label is a link to the documentation for that setting.

All of the links to Prusa documentation have been removed.
In contrast, the new settings that have been added by Bambu Lab have a working link to the wiki page.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Bambu Studio
2. Go to the Prepare tab
3. Click on Layer Height
4. Nothing happens

**Expected behavior**

It should open https://help.prusa3d.com/article/layers-and-perimeters_1748#layer-height.
I don't expect Bambu Lab to rewrite the Prusa Slicer documentation. I do expect you to not remove a large portion of the usability of the application just because you have put your name on a derived application.

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: , Windows
 - Version 1.0.0.65

bambu123, 2023-04-28:
Jumping to prusa is not good, we will add some documentation.

julie777, 2023-04-29:
Why reinvent the wheel. It is obvious that Studio is mostly Prusa Slicer, which is mostly Slic3r.
I cannot imagine that anyone would want to rewrite hundreds of pages of documentation.