bambulab/BambuStudio #1684: Settings are applied to all plates instead of the selected plate.

julie777, 2023-04-25:
**Describe the bug**

Plates are documented as having the ability to use different settings.
When I select a plate apply settings and slice it, those settings are used. 
If I then select  a different plate and change settings then slice it those settings are use. 
If I then switch between plates in preview it will show that they are both sliced with the last settings chosen.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Create two plates and put an object on each one.
2. Select the first plate and choose .20 standard and slice. 
3. Select layer height in the slice output box and see that it was sliced at .20 layer height.
4. go back to prepare and select the other plate, choose .12 fine and slice
5. select layer height in the slicer output box and you will see that the layers are .12 high
6. select the tile for the first plate
7. The plate gets re-sliced with the wrong settings
8. look at the layer height in the output box and the layer height is now .12mm

**Expected behavior**

I expect settings to be applied to the selected plate and be retained.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows]
 - Version 1.6.0.61

bambu123, 2023-04-28:
Apply presets to plate is going to be a feature.

julie777, 2023-12-13:
I have already added settings per plate as a request. It is not 9 months later, and that has not been implemented, even though it is the most needed feature for using a single project file for a complex project. Without it multiple plates are often not useful.