bambulab/BambuStudio #1685: Height Range Modifiers Not Saved in 3mf

kgearhardt, 2023-04-25:
**Describe the bug**
Height Range Modifier settings are not saved in the 3mf file.

**3mf File for This Bug**
File attached for testing 
[test-height-range-modifier.3mf.zip](https://github.com/bambulab/BambuStudio/files/11326967/test-height-range-modifier.3mf.zip)

**To Reproduce**
Select the object and create a Height Range Modifier for a range from 1 to 2mm.  Change the layer height in that range to 0.1mm.  
Save the project.

**Expected behavior**
Restart Bambu Studio and open the previously saved 3mf file.  The Height Range Modifier settings will be gone.

**Desktop (please complete the following information):**
 - OS: macOS Ventura 13.3.1

bolsoncerrado, 2023-04-26:
Did you test this in the new 1.6.0 Beta?

LiZ-Li-BBL, 2023-05-08:
We've fixed this issue.

closed 2023-05-08 by LiZ-Li-BBL

kgearhardt, 2023-05-08:
Thank you!Sent from my iPhone