bambulab/BambuStudio #1686: "Only one wall on top surfaces" should work for all top surfaces, not just the topmost one.

RebeccaStevens, 2023-04-26:
**Is your feature request related to a problem? Please describe.**
I want to use the "Only one wall on top surfaces" option to only have one wall on an inset top surface.

**Describe the solution you'd like**
I want to use the "Only one wall on top surfaces" should work for all top surfaces, not just the topmost one.

**Describe alternatives you've considered**
As a work around I can add a modifier geometry over the layer in question that changes the wall count to 1. However, this is very tedious.

**Additional context**
Screenshot showing only one wall on topmost layer, but not on inset top layers:
[image]

bolsoncerrado, 2023-04-26:
I've never understood the "ONLY ONE TOP WALL" feature... isn't this making the infill "noticed" or "seen thru"?

DzzD, 2023-04-26:
> I've never understood the "ONLY ONE TOP WALL" feature... isn't this making the infill "noticed" or "seen thru"?

No this make the part look nicer especially with ironing.

In addition to this request it should also works with Arachne wall mode and for bottom

bolsoncerrado, 2023-04-26:
How does it make nicer? Still dont get the point, sorry!

RebeccaStevens, 2023-04-26:
> How does it make nicer? Still dont get the point, sorry!

If you notice in the image I provided, the yellow surface has a bunch of walls surrounding the blue part. These walls look kind of ugly once printed.
While the blue part only has a single wall on its top surface, which looks a lot nicer.

bolsoncerrado, 2023-04-26:
Yes I can understand that, what worries me is "infill bleeding", if anything.

Thanks tho.

DzzD, 2023-04-26:
> Yes I can understand that, what worries me is "infill bleeding", if anything.

Don't worry this is an old feature of Super Slicer, you will love, actually with modifier it is already doable but tedious ;)

QingZhangBambu, 2023-05-04:
if you don't mind. Could you provide your 3mf files. I will check that.

RebeccaStevens, 2023-05-04:
I didn't save the original 3mf file, and I can't seem to recreate this anymore.
I'm not getting any walls around the icon at all now.

QingZhangBambu, 2023-06-13:
do you use OrcaSlicer? it's a bug of the arachne generator. BambuSlicer will add top one wall of Arachne soon

QingZhangBambu, 2023-08-08:
we implement only one wall top function for Arachne. you can get full functionality software after we released 1.7 hotfix2

closed 2023-08-09 by QingZhangBambu