bambulab/BambuStudio #1688: v1.6.0 Public Beta AMS selection wont let me select generic Silk PLA

cobyn, 2023-04-26:
**Describe the bug**
Generic Silk PLA is no longer an option in AMS filament options

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Device'
2. Click on an AMS Slot
3. Click on Filament DropDown
5. See that there is no option for silk pla filament

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

v1.5.0 Hotfix1
[image]

v1.6.0 Public Beta
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version v1.6.0 Public Beta

bambu123, 2023-04-28:
[image]

Have you selected it here.

cobyn, 2023-04-28:
> [image]
> 
> Have you selected it here.

Yes this only effected filament selection of the slicing setting not the AMS

bambu123, 2023-05-08:
@cobyn Does it exists in filament presets list?
[image]

cobyn, 2023-05-08:
Yes it shows up in filament section of the slicer, but it not available in the AMS filament type selection. It was available in v1.5

StoneLiBambulab, 2023-05-22:
this is a bug, the filament preset "Bambu PLA Silk" has the same filament_id with "General PLA Sillk".
the filament_id is the unique id for a filament preset, we will change a filament_did for General PLA Silk.

Thanks！

Stone Li

QingZhangBambu, 2023-06-13:
we will fix it soon

QingZhangBambu, 2023-08-01:
That has been fixed on version 1.7, you can update your software to get full functionality.

closed 2023-08-08 by QingZhangBambu