bambulab/BambuStudio #1689: Control tab does not work as expected

julie777, 2023-04-26:
**Describe the bug**

When I use control-tab, Bambu Studio changes from preview to device. This is what I would expect. Most tabbed applications work this way (browsers, editors, etc.)

However, control-shift-tab does not select the tab to the left of the current tab as expected. Instead it changes the view from the device left nav bar.

I just realized while writing this and taking the steps to reproduce the issue, that it is not consistent. 

If I open the app and use ctrl-tab and ctrl-shift-tab it does cycle through tabs correctly in both directions. 

Sometimes it starts changing the "view" of the device tab instead of changing the application tab. 
  - ctrl-tab functionality should always work on the application tabs.
  - navigation choices in the sidebar should be changed with the up and down arrow.
  - I still do not know why sometimes the mode shifts when I am looking at the device tab. I am not selecting anything on the device view. I just ctrl-tab from preview to device and then try to shift back.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. as above
2. 
**Expected behavior**

Tab switching shortcuts should work in the standard way that users have come to expect from using browsers and other tabbed applications.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:  Windows]
 - Version 1.6.0.65

walterwongbbl, 2023-05-18:
Hi.
We will optimize the UI control related code.

bambu123, 2023-07-11:
Close as fixed in the 1.7 beta version.

closed 2023-07-11 by bambu123