bambulab/BambuStudio #1692: No error message on cloud services unavailability

tubbybtch, 2023-04-27:
**Describe the bug**
When submitting a print job with Bambu Cloud Services down, there is no error message...it just throws the job into the ether.

**Expected behavior**
A concise, descriptive error dialog should display (i.e. "Bambu Cloud Services unavailable...would you like to submit by LAN instead?")  The job should NEVER queue on the cloud, as printing pileup will occur.

StoneLiBambulab, 2023-06-19:
Hi @tubbybtch ,
we added more info and a link to https://status.bambulab.com/  when bambu cloud service is down.

Thanks!
Stone Li

tubbybtch, 2023-06-19:
Fantastic...

B

On 6/18/2023 8:08 PM, Stone Li wrote:

Hi @tubbybtch<https://github.com/tubbybtch> ,
we added more info and a link to https://status.bambulab.com/ when bambu cloud service is down.

Thanks!
Stone Li

—

CAUTION This email originated from outside your organization. Please exercise caution when clicking any links or opening attachments.

Brian A. Kamp, MS Senior Lecturer (Retired) Eastern Washington University Dept. of Computer Science

closed 2024-01-19 by SaltWei