bambulab/BambuStudio #1695: [FEATURE REQUEST] Use SHIFT to select more than one object (and in general...)

bolsoncerrado, 2023-04-27:
**Is your feature request related to a problem? Please describe.**
MacOS Ventura:
Problem is that almost every software out there uses the SHIFT key to "add" (files, select more than one, etc.) but you decided to use the COMMAND key instead. Why?

**Describe the solution you'd like**
Perhaps a user preference setting to behave like the rest of softwares out there?

**Describe alternatives you've considered**
Can't.

bambu123, 2023-08-21:
We use the shift modifier for select objects by rectange, work with mouse operation.
In object list,  the shift modifier is used for range selecting.

bolsoncerrado, 2023-08-21:
Nevermind about this one. THank you very much!

closed 2023-08-21 by bolsoncerrado