bambulab/BambuStudio #1698: Fillament Selection

neilhigbee, 2023-04-27:
The new update to allow us to add filament options to the external spool holder like you can with the AMS

I would like to request the a feature to be added that allows us to select it when slicing so it uses the settings set for that fillament
As it still only lets us select the filaments in the AMS for objects in the slicer

shin-kamui, 2023-05-06:
Yes, currently syncing the ams filament list doesn't include the external spool if configured.  As a long term feature enhancement (potentially already being worked on based on the last update), could you confirm the possibility of adding the external spool to the material list and enable slicing with a combination of the AMS and external spool?   A pause and message would be required when the user needs to load or unload the external spool (maybe via push message in the app?).  Properly oriented / designed prints could really benefit from automating the workflow necessary for multi material prints with materials not supported in the AMS.  (like a phone case with a hard back plate with a custom design with tpu or flexible pla sides)

lanewei120, 2023-07-25:
hello

as the material on the external spool can not be detected by printer
we can not sync this information to slicer just the same as the filaments in AMS