bambulab/BambuStudio #1700: Allow changing not used AMS slots settings while printing

fadnincx, 2023-04-28:
# Use case / current behavior
I seldom do multi-color prints with the AMS, but I often preload the AMS with the filaments I'll use in the next prints during an already running print job. But if a print job is running, you can not change the filament slot settings after you changed the roles. You get the following message:

[image]
So I have to remember what filament I loaded in which slot until after the print.

You may ask yourself why I do this? Since I travel a lot and am only physically at the printer every few day to once a week. I have someone who can swap build plates for me. But I don't want to burden this person with changing filaments. Once the build plate has been changed, I start the jobs remotely and select the desired filament from the AMS.

# Suggested behavior
Allow changing setting for all slots not used in the current print job also during a job. I don't see a reason on why it should not be possible.

bambu123, 2023-05-04:
We'll have a discussion.

bani6809, 2023-08-01:
Why do I have to wait 4 hours for current ASA print using slot 2 to complete, just so I can change unused slot 4 from PETG to PC? 😟

please support this.