bambulab/BambuStudio #1701: Linux: Start-Up Error index.html:1:9: ReferenceError: Can't find variable: __wxOut0

norpol, 2023-04-28:
**Describe the bug**

Starting up BambuStudio (and some forks see similar previously created issue  issue https://github.com/SoftFever/OrcaSlicer/issues/442 which shows that the bug occurs for a variety of users not dependent on distribution, window manager, GPU driver, ...) sometimes shows up an error like this one, although everything appears to be functional after closing the dialog.

 > `Error running JavaScript: file:///nix/store/m0aq9bm6rn2ym3r66z21cp57prlvb8v6-bambu-studio-softfever-1.4.5/share/BambuStudio/web/homepage/index.html:1:9: ReferenceError: Can't find variable: __wxOut0`

or

> `"Error running Javascript: file:///tmp/.mount_BambuSKzd8iH/resources/web/homepage/index.html?lang=en_GB:1:9: ReferenceError: Can't find variable: __wxOut0"`

if you are using the AppImage of forks https://github.com/SoftFever/BambuStudio-SoftFever/pull/86 but also the 1.6.0 Beta Image for Fedora https://github.com/bambulab/BambuStudio/issues/12#issuecomment-1511424730 

**To Reproduce**
Open bambu-studio several times, 

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Version: v1.6.0 Fedora Beta Image

deftdawg, 2023-05-01:
The origin of this message is this line within bambu-studio is `deps/build/destdir/usr/local/include/wx-3.1/wx/private/jsscriptwrapper.h` on line 37:
```cpp
        m_outputVarName = wxString::Format(wxASCII_STR("__wxOut%i"), (*runScriptCount)++);
```

It was replaced in wx-widgets about 4 commits to the file ago...  But it used to be here:
https://github.com/wxWidgets/wxWidgets/blob/c86bcf962d014fb1f17a04a26da584c746cad1f4/include/wx/private/jsscriptwrapper.h#L37

bambu123, 2023-05-04:
We have fixed it in version 1.6.

bambu123, 2023-05-08:
The reason is when 'AddScriptMessageHandler', it keeps ui message loops in 'RunScript', and another 'AddScriptMessageHandler' is called which uses the same temporary variable "__wxOut0".

We made them sequential, but they seem not to be fixed fully.
The fact is the ScriptMessageHandler only needs to be added only once for multiple WebViews.

deftdawg, 2023-05-16:
> @deftdawg Does it happen every start-up or occasionally?

Everytime, on start up there is pop up about `playbin` not found followed by this pop-up about `wxOut0`

bambu123, 2023-05-16:
@deftdawg Could you please send me the logs from the menu 'Help > Open Configurataion Folder'?

nmcbride, 2023-05-16:
Happens to me as well when I startup.

[image]

bambu123, 2023-05-16:
@nmcbride Could you please send me the logs from the menu 'Help > Open Configurataion Folder'?

nmcbride, 2023-05-16:
@bambu123 Uploaded.

deftdawg, 2023-05-16:
Here are my logs from running the latest Ubuntu app-image release on NixOS
[debug_.tar.gz](https://github.com/bambulab/BambuStudio/files/11484321/debug_.tar.gz)

theanswriz42, 2023-05-28:
Same issue for me on 1.6.3 (not beta):

`steve@pluto:~/Downloads$ ./OrcaSlicer_ubu64.AppImage 
[2023-05-28 11:21:42.262259] [0x00007f4151ac1980] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/steve/.config/OrcaSlicer/plugins/libbambu_networking.so
after dlopen, network_module is 0x55d252cc9cd0

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.764: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.764: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): GLib-GObject-CRITICAL **: 11:21:43.765: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.765: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.766: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.766: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): GLib-GObject-CRITICAL **: 11:21:43.766: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.766: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:22196): GLib-GObject-CRITICAL **: 11:21:43.768: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.768: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:22196): GLib-GObject-CRITICAL **: 11:21:43.770: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.770: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.777: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.777: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): GLib-GObject-CRITICAL **: 11:21:43.777: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.777: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.780: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.780: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): GLib-GObject-CRITICAL **: 11:21:43.780: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.780: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.898: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.903: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.903: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.905: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.905: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.905: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.905: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.905: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.905: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.919: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:22196): Gtk-CRITICAL **: 11:21:43.921: gtk_widget_set_size_request: assertion 'width >= -1' failed
[... 462 lines cut ...]

(orca-slicer:22196): Gtk-CRITICAL **: 11:27:41.907: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:22196): GLib-GObject-CRITICAL **: 11:27:41.911: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:22196): Gtk-CRITICAL **: 11:27:41.911: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:22196): GLib-GObject-CRITICAL **: 11:27:41.911: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:22196): Gtk-CRITICAL **: 11:27:41.911: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:22196): GLib-GObject-CRITICAL **: 11:27:41.929: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:22196): Gtk-CRITICAL **: 11:27:41.929: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:22196): GLib-GObject-CRITICAL **: 11:27:41.930: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:22196): Gtk-CRITICAL **: 11:27:41.930: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:22196): GLib-GObject-CRITICAL **: 11:27:41.931: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:22196): Gtk-CRITICAL **: 11:27:41.931: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
`

[debug_network_Sun_May_28_11_22_06.log](https://github.com/bambulab/BambuStudio/files/11584983/debug_network_Sun_May_28_11_22_06.log)

bambu123, 2023-05-29:
Will be fixed in the next version.

closed 2023-07-11 by bambu123