bambulab/BambuStudio #1703: `Unable to load shaders` error popup while trying to slice a stl file using the cli

humsie, 2023-04-29:
**Describe the bug**
While trying to slice a stl file using the cli as described in https://github.com/bambulab/BambuStudio/wiki/Command-Line-Usage multiple `Unable to compile vertex shader of shader program` errors are thrown saying the executable can't find/load it's shaders.
And these errors trigger a GUI popup requiring the user to click on OK before the command line execution continues which breaks the cli usefulness in automation.

Context: Trying to automate generation of Color/Filament samples containing filament details like brand/name etc

**3mf File for This Bug**
Any stl file will do

**To Reproduce**
Tried to run BambuStudio executable from different locations, including the shaders folder, but non worked
```
cd /Applications/BambuStudio.app/Contents/MacOS/

./BambuStudio \
--orient --arrange 1 \
--curr-bed-type "Cool Plate" \
--load-settings "../Resources/profiles/BBL/machine/Bambu Lab X1 Carbon 0.4 nozzle.json;../Resources/profiles/BBL/process/0.20mm Standard @BBL X1C.json" \
--load-filaments "../Resources/profiles/BBL/filament/Generic PLA.json" \
--export-3mf /path/to/output.3mf \
--slice 0 \
/path/to/pla_sunlu-silk_rainbow_200.stl

```

Output:
```
[2023-04-29 11:41:20.919127] [0x00007ff844b72340] [trace]   Initializing StaticPrintConfigs
                                      overhang, bottom, bothull, contour, A_laf, A_prj, unprintability
orientation:-0.0000 -0.0000  1.0000, cost:3.2,	3527.2,	2362.5,	237.6,	0.3,	0.0,	0.0
orientation:-0.0000 -0.0000 -1.0000, cost:1842.8,	760.6,	2362.5,	110.3,	0.4,	0.0,	2.6
orientation:-0.0000  1.0000 -0.0000, cost:407.3,	235.1,	158.7,	61.3,	83.5,	0.0,	2.2
orientation:-0.0000 -1.0000  0.0000, cost:413.6,	228.7,	158.7,	60.5,	83.5,	0.0,	2.3
orientation: 1.0000 -0.0000 -0.0000, cost:313.1,	66.5,	45.3,	32.6,	102.2,	0.0,	5.7
orientation:-1.0000  0.0000 -0.0000, cost:313.4,	66.5,	45.3,	32.6,	102.2,	0.0,	5.7
orientation:-0.9613  0.2754 -0.0000, cost:352.6,	5.8,	6.3,	9.6,	72.0,	0.0,	56.4
orientation:0.9630 0.2697 0.0000, cost:351.1,	5.9,	5.8,	9.7,	78.5,	0.0,	55.5
orientation:0.9469 0.3214 0.0000, cost:347.9,	6.5,	4.3,	10.2,	81.6,	0.0,	51.8
orientation: 0.9671 -0.2544 -0.0000, cost:356.2,	1.1,	0.6,	4.2,	80.7,	0.0,	198.7
orientation:-0.9835 -0.1811 -0.0000, cost:355.4,	1.4,	7.3,	4.7,	395.6,	0.0,	148.4
orientation: 0.9835 -0.1811 -0.0000, cost:356.0,	1.4,	7.3,	4.7,	391.1,	0.0,	148.5
orientation: 0.6594 -0.7518 -0.0000, cost:736.5,	6.5,	4.3,	10.2,	56.6,	0.0,	108.5
orientation:-0.7518  0.6594 -0.0000, cost:746.2,	6.5,	4.5,	10.2,	60.1,	0.0,	109.7
orientation:-0.7071 -0.0000  0.7071, cost:2665.3,	1.0,	0.0,	4.1,	0.2,	0.0,	1529.8
orientation:-0.0000 -0.7071  0.7071, cost:2862.2,	5.0,	0.0,	8.9,	0.0,	0.0,	539.5
orientation: 0.7071 -0.0000  0.7071, cost:2665.7,	0.0,	0.0,	0.5,	0.2,	0.0,	5782.7
orientation:-0.0000  0.7071  0.7071, cost:2866.0,	0.0,	0.0,	0.5,	0.0,	0.0,	6209.4
orientation:-0.7071 -0.7071 -0.0000, cost:739.8,	5.8,	5.8,	9.6,	65.5,	0.0,	118.0
orientation: 0.7071 -0.7071 -0.0000, cost:741.3,	5.8,	5.8,	9.6,	70.6,	0.0,	118.2
orientation: 0.7071  0.7071 -0.0000, cost:746.6,	5.8,	5.8,	9.6,	65.5,	0.0,	119.0
orientation:-0.7071  0.7071 -0.0000, cost:748.4,	5.8,	6.3,	9.6,	70.6,	0.0,	118.7
orientation:-0.7071 -0.0000 -0.7071, cost:2665.4,	0.0,	0.0,	0.0,	0.2,	0.0,	6921.9
orientation:-0.0000 -0.7071 -0.7071, cost:2865.3,	3.7,	0.0,	7.7,	0.0,	0.0,	679.3
orientation: 0.7071 -0.0000 -0.7071, cost:2665.8,	0.0,	0.0,	0.0,	0.2,	0.0,	6922.9
orientation:-0.0000  0.7071 -0.7071, cost:2865.7,	0.0,	0.0,	0.0,	0.0,	0.0,	7434.1
best:-0.000000 -0.000000  1.000000, costs:3.2,	3527.2,	2362.5,	237.6,	0.3,	0.0,	0.0
st=1, pla_sunlu-silk_rainbow_200.stl
after arrange: name=pla_sunlu-silk_rainbow_200.stl,bed_id 0, trans {139.75,115}
[2023-04-29 11:41:21.745839] [0x00007ff844b72340] [error]   Unable to compile vertex shader of shader program 'gouraud_light':
ERROR: 0:1: '' :  version '110' is not supported
ERROR: 0:2: '' :  #version required and missing.
ERROR: 0:18: 'varying' : syntax error: syntax error

[2023-04-29 11:41:21.746168] [0x00007ff844b72340] [error]   Unable to compile vertex shader of shader program 'cali':
ERROR: 0:1: '' :  version '110' is not supported
ERROR: 0:2: '' :  #version required and missing.

[2023-04-29 11:41:21.746466] [0x00007ff844b72340] [error]   Unable to compile vertex shader of shader program 'printbed':
ERROR: 0:1: '' :  version '110' is not supported
ERROR: 0:2: '' :  #version required and missing.
ERROR: 0:3: 'attribute' : syntax error: syntax error

[2023-04-29 11:41:21.746770] [0x00007ff844b72340] [error]   Unable to compile vertex shader of shader program 'gouraud_light_instanced':
ERROR: 0:1: '' :  version '110' is not supported
ERROR: 0:2: '' :  #version required and missing.
ERROR: 0:18: 'attribute' : syntax error: syntax error

[2023-04-29 11:41:21.747061] [0x00007ff844b72340] [error]   Unable to compile vertex shader of shader program 'toolpaths_lines':
ERROR: 0:1: '' :  version '110' is not supported
ERROR: 0:2: '' :  #version required and missing.
ERROR: 0:3: 'varying' : syntax error: syntax error

[2023-04-29 11:41:21.747485] [0x00007ff844b72340] [error]   Unable to compile vertex shader of shader program 'gouraud':
ERROR: 0:1: '' :  version '110' is not supported
ERROR: 0:2: '' :  #version required and missing.
ERROR: 0:37: 'varying' : syntax error: syntax error

[2023-04-29 11:41:21.747824] [0x00007ff844b72340] [error]   Unable to compile vertex shader of shader program 'variable_layer_height':
ERROR: 0:1: '' :  version '110' is not supported
ERROR: 0:2: '' :  #version required and missing.
ERROR: 0:21: 'varying' : syntax error: syntax error

[2023-04-29 11:41:21.748137] [0x00007ff844b72340] [error]   Unable to compile vertex shader of shader program 'mm_contour':
ERROR: 0:1: '' :  version '110' is not supported
ERROR: 0:2: '' :  #version required and missing.

[2023-04-29 11:41:21.748438] [0x00007ff844b72340] [error]   Unable to compile vertex shader of shader program 'mm_gouraud':
ERROR: 0:1: '' :  version '110' is not supported
ERROR: 0:2: '' :  #version required and missing.
ERROR: 0:11: 'varying' : syntax error: syntax error

[2023-04-29 11:41:21.748698] [0x00007ff844b72340] [error]   Unable to compile vertex shader of shader program 'outline':
ERROR: 0:1: '' :  version '110' is not supported
ERROR: 0:2: '' :  #version required and missing.
ERROR: 0:3: 'attribute' : syntax error: syntax error

... Waits for OK click in Gui POPUP ...

[2023-04-29 11:36:30.774136] [0x00007ff844b72340] [error]   can not get shader for rendering thumbnail
```

GUI Popup contents:
```
Error Loading Shaders

Unable to load shaders:
gouraud_light
cali
printbed
gouraud_light_instanced
toolpaths_lines
gouraud
variable_layer_height
mm_contour
mm_gouraud
outline
```

**Expected behavior**
1. A CLI slicing application should not require GUI input before continuing
2. The application should know how to load it's shaders 

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: 16.4
 - Bambu Studio: 1.5.0.61

lanewei120, 2023-05-04:
thanks for reporting this issue

we will add a solution to not popup error window
however, for the shader loading fail, we have no idea now

lanewei120, 2023-08-05:
already close the popup

closed 2023-08-05 by lanewei120