bambulab/BambuStudio #1704: Slicing difference between CLI and GUI

humsie, 2023-04-29:
**Describe the bug**
After slicing a STL using the cli below, the baked preset in the 3mf file is very different than the selected settings.

Opening the `pla_sunlu-silk_rainbow_200.3mf` in BambuStudio the print time is `18m28s` instead of `8m3s` when selecting the `0.20mm Standard @BBL X1C` from the drop down

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Slice the STL with CLI (also see #1703):
```
cd /Applications/BambuStudio.app/Contents/MacOS/
./BambuStudio \
--orient --arrange 1 \
--curr-bed-type "Cool Plate" \
--load-settings "../Resources/profiles/BBL/machine/Bambu Lab X1 Carbon 0.4 nozzle.json;../Resources/profiles/BBL/process/0.20mm Standard @BBL X1C.json" \
--load-filaments "../Resources/profiles/BBL/filament/Generic PLA.json" \
--export-3mf /path/to/pla_sunlu-silk_rainbow_200.3mf \
--slice 0 \
/path/to/pla_sunlu-silk_rainbow_200.stl
```

2. Open resulting 3mf in BambuStudio
3. Hit `Slice Plate`
4. See Model printing time of `18m28s`
5. Select `0.20mm Standard @BBL X1C` preset 
6. See multiple settings change for example the line-width values
3. Hit `Slice Plate`
4. See Model printing time of `8m3s`

**Expected behavior**
No difference between the `project-inside presets` and the `system preset` used for slicing

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: 16.4
 - Bambu Studio: 1.5.0.61

lanewei120, 2023-05-04:
hello, the file "../Resources/profiles/BBL/machine/Bambu Lab X1 Carbon 0.4 nozzle.json" is not a full config file
it inherits from fdm_bbl_3dp_001_common.json, it only includes some differences to the parent
so it is not suitable to pass it as a settings to CLI

furthermore, our CLI slicing is not suitable to slice stl file, it is more suggested to slice 3mfs

NathaelB, 2023-06-22:
hello, we can dockerize the BambuStudio application, in order to slice a stl file with our own process and then retrieve the 3mf file automatically.

The problem I've encountered is that, when I give a `process.json` config file, it doesn't take the given parameters into account. At first, I simply gave a file with the inherits, then I switched to the method of giving the entire config on a single file, so the `initial_layer_speed` normally given for 50 is 30 when the application slice my file.

lanewei120, 2023-06-25:
@NathaelB 
so what is your slicing commands now?

and could you also provide your configs passed to the slicing command

theyagizsina, 2025-04-24:
> hello, the file "../Resources/profiles/BBL/machine/Bambu Lab X1 Carbon 0.4 nozzle.json" is not a full config file it inherits from fdm_bbl_3dp_001_common.json, it only includes some differences to [...]

I encountered the same problem, I am trying to make a web app to make a price calculation with the data after the stl upload and slice on the web side. I can slice with 3mf without any problem, but it does not give the correct result due to inherits with machine profiles etc.

I want to learn how Bambu Studio combines these in the background, I think I can write a CLI wrapper based on this, it would be great if I could learn where the slicer combines this data in itself.

Translated with [DeepL.com](https://www.deepl.com/?utm_campaign=product&utm_source=web_translator&utm_medium=web&utm_content=copy_free_translation) (free version)