bambulab/BambuStudio #1705: Multi material support layer shift problem

ByTechLab, 2023-04-29:
**Describe the bug**
After switching the "Support interface" to a different material than the "Support base" stragne things happen - missing layers, shifted layers and mirrored layers (see screens below).

**3mf File for This Bug**
[Fan_Stator_Casing_and_Support_With_Supports.zip](https://github.com/bambulab/BambuStudio/files/11359081/Fan_Stator_Casing_and_Support_With_Supports.zip)
[HP_Core_Casing.zip](https://github.com/bambulab/BambuStudio/files/11359069/HP_Core_Casing.zip)

**To Reproduce**
Steps to reproduce the behavior:
Slice with the same filament for the support base and interface and with different filaments:
[image]

**Expected behavior**
The interface material should be placed between the support base and the model. and there should be no gaps in the layers.

**Screenshots**
The same filament for support base and interface:
[image]

Different filaments:
[image]
[image]
[image]

[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 Home
[image]
[image]

ByTechLab, 2023-04-30:
It seems to be connected with the "Z hop when retracting" variable.
When set to 0 I have all of the issues described above.
When set to anything else > 0 all issues disappear.

tangketan, 2023-05-05:
The shifted support interfaced is a fixed bug. Please try the latest version 1.6.2 from the official site. 
The last picture you gave indicates that the thin walls are not sliced correctly. You could try Arachne
[image]

closed 2023-05-05 by ByTechLab